skycoin/hardware-wallet #211: Deployment keys

olemis, 2019-05-03:
**Feature description**

Add keys for Travis `deploy`

**Is your feature request related to a problem? Please describe.**
Auth failure during Travis deploy since keys not generated for upstream repository

**Describe the solution you'd like**
Secure keys in Travis script for deployment tokens (see details below)

**Additional context**
#210 and #183 create in spite of writing code to release artifacts that need to be signed 

**Possible implementation**

- [ ] Add secure key for Github Releases 
- [ ] Add secure key for S3 deployment

stdevAlDen, 2019-05-08:
> **Feature description**
>
> Add keys for Travis `deploy`
>
> **Is your feature request related to a problem? Please describe.**
> Auth failure during Travis deploy since keys not generated for upstream [...]

https://github.com/skycoin/hardware-wallet/pull/214#issuecomment-489680610

gz-c, 2019-05-09:
Setup s3 keys in travis

stdevAlDen, 2019-05-09:
> Setup s3 keys in travis

excuse me @gz-c  did you set up a suitable key for the github releases too?
- https://github.com/skycoin/hardware-wallet/blob/ffd6fcdaf1d89a0877372f6549cd8a4035a49760/.travis.yml#L63
 - https://github.com/skycoin/hardware-wallet/blob/ffd6fcdaf1d89a0877372f6549cd8a4035a49760/.travis.yml#L76

gz-c, 2019-05-26:
I set the `AWS_ACCESS_KEY_ID`, `AWS_BUCKET`, `AWS_REGION` and `AWS_SECRET_ACCESS_KEY` variables in travis.  I did not use travis.yml's encrypted `secure` field.  Refer to github.com/skycoin/skycoin for the way we use these keys to upload the artifacts

olemis, 2019-05-26:
> I set the `AWS_ACCESS_KEY_ID`, `AWS_BUCKET`, `AWS_REGION` and `AWS_SECRET_ACCESS_KEY` variables in travis

@gz-c thanks . Nonetheless it seems Github Releases token still missing . Could you please add it ? Can't do it myself .

PR #241 (2019-06-02): [ci] fixes #211 - Do not deploy to Github Releases
Fixes #211

Changes:	
- Do not deploy to Github Releases

Does this change need to mentioned in CHANGELOG.md?	
no

Requires testing	
no
commit 46ef0d34 (2019-06-02): Merge pull request #241 from simelo/olemis_t211_rm_deploy_releases

[ci] fixes #211 - Do not deploy to Github Releases
closed 2019-06-02 by olemis