skycoin/hardware-wallet #215: Error generating address

stdevAlDen, 2019-05-06:
**Describe the bug**
Error in response sequence for `fsm_msgSkycoinAddress` 

**Environment information:**
Any

**Steps to Reproduce**
Run tests for hw-go

**Actual behavior**
Sending a [response by mistake](https://github.com/skycoin/hardware-wallet/blob/160614a3ec458da4e3ca238ecec9f344f54416f9/tiny-firmware/firmware/fsm.c#L393) after a [valid one actually send](https://github.com/skycoin/hardware-wallet/blob/160614a3ec458da4e3ca238ecec9f344f54416f9/tiny-firmware/firmware/fsm.c#L376)

**Expected behavior**
Fix the protocol sequence.
Run integration tests from `go` and `js` libraries after PR.

**Additional context**
This is blocking this https://github.com/skycoin/hardware-wallet-go/pull/121

**Possible implementation**
As `go` library is currently running integration tests from firmware analyza how to avoid fall down in a circular dependency for failing tests.

therealssj, 2019-05-06:
```

void fsm_msgSkycoinAddress(SkycoinAddress* msg)
{
    RESP_INIT(ResponseSkycoinAddress);
    char* failMsg = NULL;
    ErrCode_t err = msgSkycoinAddressImpl(msg, resp);
    switch (err) {
    case ErrUserConfirmation:
        layoutAddress(resp->addresses[0]);
        if (!protectButton(ButtonRequestType_ButtonRequest_ProtectCall, false)) {
            err = ErrActionCancelled;
            break;
        }
        // fall through
    case ErrOk:
        msg_write(MessageType_MessageType_ResponseSkycoinAddress, resp);
        break;
    case ErrPinRequired:
        failMsg = _("Expected pin");
        break;
    case ErrTooManyAddresses:
        failMsg = _("Asking for too much addresses");
        break;
    case ErrMnemonicRequired:
        failMsg = _("Mnemonic required");
        break;
    case ErrAddressGeneration:
        failMsg = _("Key pair generation failed");
        break;
    default:
        fsm_sendResponseFromErrCode(err, NULL, failMsg);
        break;
    }
    layoutHome();
}
```
imo that is the correct implementation

therealssj, 2019-05-06:
@stdevAlDen i think it would be good to go through other functions and look for similar bugs ( if this indeed is a bug )

stdevAlDen, 2019-05-06:
> ```
> 
> void fsm_msgSkycoinAddress(SkycoinAddress* msg)
> {
>     RESP_INIT(ResponseSkycoinAddress);
>     char* failMsg = NULL;
>     ErrCode_t err = msgSkycoinAddressImpl(msg, resp);
>     switch (err) {
>     case ErrUserConfirmation:
>         layoutAddress(resp->addresses[0]);
>         if (!protectButton(ButtonRequestType_ButtonRequest_ProtectCall, false)) {
>             err = ErrActionCancelled;
>             break;
>         }
>         // fall through
>     case ErrOk:
>         msg_write(MessageType_MessageType_ResponseSkycoinAddress, resp);
>         break;
>     case ErrPinRequired:
>         failMsg = _("Expected pin");
>         break;
>     case ErrTooManyAddresses:
>         failMsg = _("Asking for too much addresses");
>         break;
>     case ErrMnemonicRequired:
>         failMsg = _("Mnemonic required");
>         break;
>     case ErrAddressGeneration:
>         failMsg = _("Key pair generation failed");
>         break;
>     default:
>         fsm_sendResponseFromErrCode(err, NULL, failMsg);  // <--- moved this line up
>         break;
>     }
>     layoutHome();
> }
> ```
> 
> imo that is the correct implementation

@therealssj I can see that you miss a little detail in this propoassal, in any case, what do you think about for this builds/test download hw-go in travis and run hw-go integration tests against this emulator.
If you look at how the bg become, the curren firmware test battery is not able to catch the bug.

olemis, 2019-05-06:
@stdevAlDen for firmware , no ...

olemis, 2019-05-06:
there's indeed a bug

> imo that is the correct implementation

... but this change will not fix it , indeed it introduces a new bug

therealssj, 2019-05-06:
> there's indeed a bug
> 
> > imo that is the correct implementation
> 
> ... but this change will not fix it , indeed it introduces a new bug

that proves that i can't write c :D

therealssj, 2019-05-06:
i agree with @olemis the tests shudnt be run in firmware

olemis, 2019-05-07:
fixed in 2ffb2af

PR #216 (2019-05-07): Gen address bug ref #215
Fixes #215 

Changes:
- Fxi bug in address gen
- Run hardware-wallet-go integrations tests against emulator.

Does this change need to mentioned in CHANGELOG.md?
no

Requires changes in protobuf specs?
no

Requires testing
no

Comments about testing , should you have some
commit 2ffb2af4 (2019-05-07): Merge pull request #216 from simelo/stdevAlDen_t215_gen_address_bug

Gen address bug fixes #215
closed 2019-05-07 by olemis