skycoin/hardware-wallet #219: Bug in msgSkycoinCheckMessageSignatureImpl

stdevAlDen, 2019-05-07:
**Describe the bug**
When calling [`msgSkycoinCheckMessageSignatureImpl`](https://github.com/skycoin/hardware-wallet/blob/2ffb2af49c8100dbca6645e8ff9847f1544cf3ba/tiny-firmware/firmware/fsm.c#L297) the user should see some relevant info in the screen according to this code:
https://github.com/skycoin/hardware-wallet/blob/2ffb2af49c8100dbca6645e8ff9847f1544cf3ba/tiny-firmware/firmware/fsm_impl.c#L228-L231

But this message will be overwritten immediately by a [`layoutHome`](https://github.com/skycoin/hardware-wallet/blob/2ffb2af49c8100dbca6645e8ff9847f1544cf3ba/tiny-firmware/firmware/fsm.c#L303) call

**Environment information:**

Any

**Steps to Reproduce**
Just follow the logic in source code.

**Actual behavior**
the layout home override the previos info in the screen

**Expected behavior**
TODO @olemis & @therealssj please help me to define the desired behavior

**Additional context**
None

**Possible implementation**
TODO

therealssj, 2019-05-07:
I don't think the user needs to see an error message here. 
The function should just return an error message and the desktop wallet should display an error accordingly

olemis, 2019-05-08:
fixed by 0268088

PR #222 (2019-05-08): Fix bug in skycoin sign message ref #219
Fixes #219

Changes:
- Remove not useful screen messages

Does this change need to mentioned in CHANGELOG.md?
no

Requires changes in protobuf specs?
no

Requires testing
no
commit 02680884 (2019-05-08): Merge pull request #222 from simelo/stdevAlDen_t219_fsm_msgSkycoinCheckMessageSignature_bug

Fix bug in skycoin sign message fixes #219
closed 2019-05-08 by olemis