skycoin/hardware-wallet #231: Bug in sha 256

stdevAlDen, 2019-05-14:
**Describe the bug**

**Environment information:**

- Linux 4.19.0-1-amd64 x86_64

**Steps to Reproduce**
Steps to reproduce the behavior:

```c
    uint8_t digest[SHA256_DIGEST_LENGTH] = {0};
    int seed = rand();
    compute_sha256sum((const uint8_t*)seed, digest, sizeof(seed));
```

**Actual behavior**
Received signal 11 (Segmentation fault)

**Expected behavior**
Any input message(0 <= len < 2^64) should generate a 256 bits hash.
https://crypto.stackexchange.com/a/42033

**Additional context**
https://github.com/skycoin/hardware-wallet/issues/84
https://github.com/skycoin/libskycoin/issues/34

**Possible implementation**
TODO

olemis, 2019-05-17:
> ```c
> compute_sha256sum((const uint8_t*)seed, digest, sizeof(seed));
> ```

this code is wrong it should be 

```c
compute_sha256sum((const uint8_t*)&seed, digest, sizeof(seed));
```

closed 2019-05-17 by olemis

stdevAlDen, 2019-05-18:
> > ```c
> > compute_sha256sum((const uint8_t*)seed, digest, sizeof(seed));
> > ```
> 
> this code is wrong it should be
> 
> ```c
> compute_sha256sum((const uint8_t*)&seed, digest, sizeof(seed));
> ```

@olemis great