skycoin/hardware-wallet #238: [Bootloader] Return an error if busy or in an unavailable state

Senyoret1, 2019-05-29:
**Feature description**
When sending a request to the bootloader while it is busy or in an unavailable state, it responds nothing. It should return an error indicating in some way that is it not possible to do what was requested.

**Is your feature request related to a problem? Please describe.**
I did the following using the js library:

1) I used the `devUpdateFirmware` function for updating the device and obtained this output:
[image]
Also, the device showed this:
[image]

2) While having the console in focus, I pressed `ctrl+c`. That canceled the operation in the pc, but the hardware wallet was still waiting for user input. In this state, I called `devUpdateFirmware` again a got this output:
[image]
It is important to note that the error message did not appear quickly, but after some time. This appear to indicate that the error message appeared because of a timeout and not because the device responded something in particular. Also, it was interesting that while the device was waiting for user input, the calls to the `getDevice` function also failed after some time, so the bootloader is not responding to them while waiting for input.

3) I cancelled the operation in the device, so the screen ended in this state:
[image]
After that, I called `devUpdateFirmware` again a got this output:
[image]
After the last line nothing else happened, neither on the device nor on the console, so the operation was stuck.

Testing in the console, this could appears like normal behavior, but carries some complications for the integration with a GUI. this is an example (this happened while creating the updater in the desktop wallet, using the daemon):

> If the user tries to update the firmware and by accident cancels the operation, the device screen will ask the user to reconnect the hw wallet, but the user could simply retry using the desktop wallet. In that case, the hardware wallet will do nothing but the desktop wallet will ask the user to confirm the operation in the hardware wallet. It will stay like that for 50 secons until the http connection times out.

**Describe the solution you'd like**
In the previous examples, instead of not receiving a response, an error should have been received. An example is that if I use the `devWipeDevice` function while the device is in firmware mode and call `devWipeDevice` again while de device is still waiting for user input, this is the console output:
[image]

**Describe alternatives you've considered**
The previous example about how the firmware reacts shows a possible solution, but in fact the implementation does not need to be the same. What is important is to have any kind of error, so the GUI can know that something went wrong with the operation and tell the user that the procedure was cancelled.

**Additional context**

**Possible implementation**

stdevAlDen, 2019-05-30:
@Senyoret1 an incoming command should cancel any existing one(runnin, in progress), an run the new(the second command) as a fresh device?
What do you think?

stdevAlDen, 2019-05-30:
A very strange behavior, if you run `firmwareUpdate`, cancel in the terminal (<kbd>ctrl</kbd> + <kbd>c</kbd>), unplug the device, plug it again and run `firmwareUpdate` the process start without asking for confirmation.

olemis, 2019-05-30:
> While having the console in focus, I pressed ctrl+c. That canceled the operation in the pc, but the hardware wallet was still waiting for user input. In this state, I called devUpdateFirmware again a got this output:

In this particular case , for bootloader , must cancel in the device prior to sending second `firmwareUpdate` message .

Senyoret1, 2019-05-30:
> 
> 
> @Senyoret1 an incoming command should cancel any existing one(runnin, in progress), an run the new(the second command) as a fresh device?
> What do you think?

That would also work, as the operation would be started correctly and there would not be need to return any error. However, would that work if the device is requesting the user to reconnect it and, in a more unlikely scenario, if the request was made while the device was installing a new firmware?

> 
> 
> A very strange behavior, if you run `firmwareUpdate`, cancel in the terminal (ctrl + c), unplug the device, plug it again and run `firmwareUpdate` the process start without asking for confirmation.

I tried to do that and only happened if the updating proccess had started. In that case the updating process would have deleted the firmware and the second time you call `firmwareUpdate` the device do not have a firmware, so the bootloader install the new one without asking for confirmation.

olemis, 2019-05-30:
> However, would that work if the device is requesting the user to reconnect it and, in a more unlikely scenario, if the request was made while the device was installing a new firmware?

like I said before at these two moments in time the device is unstable , no message should be neither processed nor cancelled . User must either wait or obey the instructions in device screen .

olemis, 2019-05-30:
> @Senyoret1 an incoming command should cancel any existing one(runnin, in progress), an run the new(the second command) as a fresh device?

In firmware (most of the time) yes .

olemis, 2019-05-30:
> > @Senyoret1 an incoming command should cancel any existing one(runnin, in progress), an run the new(the second command) as a fresh device?
> 
> In firmware (most of the time) we can assume this is the case .

All other sync should happen in software , e.g. an Angular service aware of in progress operation on the device displays a confirmation box for cancellation before sending new message.

olemis, 2019-05-30:
at least for v1

Senyoret1, 2019-05-30:
> 
> 
> > While having the console in focus, I pressed ctrl+c. That canceled the operation in the pc, but the hardware wallet was still waiting for user input. In this state, I called devUpdateFirmware again a got this output:
> 
> In this particular case , for bootloader , must cancel in the device prior to sending second `firmwareUpdate` message . Current behavior is correct , as far as I can tell .

If it is expected that the user should cancel the operacion in the device before calling `firmwareUpdate` again, the fact that the device does not try to install the new firmware again is ok. The problem is that the second call to `firmwareUpdate` should return an error, as it happens with the firmware, as described in the "Describe the solution you'd like" section. The problem is that not having an error response left the GUI without knowing what to do.

The problem is that there are various reasons why an operation could be requested before finishing the previous one, and not all the time is the users fault. The main example is when the GUI cancels automatically the operation because a http timeout but the device does not know that, because the bootloader does not have an instruction for cancelling the previous operation. In the firmware it is not a big deal because it is possible to cancel the previous procedure.

olemis, 2019-05-30:
> The main example is when the GUI cancels automatically the operation because a http timeout but the device does not know that, because the bootloader does not have an instruction for cancelling the previous operation.

I assume the scenario is downloading firmware binary via HTTP for flashing device firmware ... or close to that . If that is the case `firmwareUpdate` should be invoked once HTTP download finished , i.e. no HTTP transfer happening in parallel with flashing .

```js
downloadFirmware(url).then((bytes) => devFirmwareUpdate(bytes))
```

Senyoret1, 2019-05-30:
> 
> 
> > The main example is when the GUI cancels automatically the operation because a http timeout but the device does not know that, because the bootloader does not have an instruction for cancelling the previous operation.
> 
> I assume the scenario is downloading firmware binary via HTTP for flashing on device ... or close to that . If that is the case `firmwareUpdate` should be invoked once HTTP download finished , i.e. no HTTP transfer happening in parallel with flashing .
> 
> ```js
> downloadFirmware(url).then((bytes) => devFirmwareUpdate(bytes))
> ```

That is not the case. In fact, it is not possible to do it that way because it is needed to download the binary before calling `devFirmwareUpdate`, as the binary must be send as a param to `devFirmwareUpdate`. The thing is more ike this:

When the user uses the firmware updater in the desktop wallet, it checks what the lastest firmware version is and download the binary. After that, it sends the binary to the `/api/v1/firmware_update` API endpoint of the daemon and the devices ask the user for confirmation. When that happens, the user has 50 seconds to confirm the operation on the device or the http request made to the daemon for updating the firmware is cancelled due to a timeout (many browsers have a mandatory and unconfigurable timeout, so it is not possible to avoid the operation to be cancelled after some time). If the operation is cancelled due to timeout, there are two possible problems: the first one is that the user could try to continue the operation in the device and it will get stuck (for some reason the firmware is no installed if the desktop code is not longer waiting just when the previous firmware is deleted, but that is another unrelated issue), and the second one is that the user could try to start the operation again in the desktop wallet, which brings as a consequence the problem described in this issue.

Also, that is not the only possible scenario. Another caase is the one add added in the original issue:

> If the user tries to update the firmware and by accident cancels the operation, the device screen will ask the user to reconnect the hw wallet, but the user could simply retry using the desktop [...]

Both cases are pressent in https://github.com/skycoin/skycoin/pull/2372

Senyoret1, 2019-05-30:
None of those scenarios is critical, as the user can solve them by reconnecting the device, but are anoyances in the UX. So, there is not really needed to solve them if it is too complicated for the next release, but it would be nice as it would help improve the UX.

jdknives, 2019-05-30:
I understand that those things might be annoying for the user, but if it can be fixed by reconnecting, that solutionis simple enough so that we can leave it for now as it is. Please do not spend time on this @olemis

jdknives, 2019-05-30:
@stdevAlDen this is not part of Milestone 1. The comments here indicate that this is not critical for release as the user can unplug and reconnect to fix any issue even if this is not the optimal way to go.

jdknives, 2019-05-30:
I unassigned it from Milestone 1 for now. Let me know if there is something I misunderstood.

Senyoret1, 2019-05-30:
> 
> 
> I unassigned it from Milestone 1 for now. Let me know if there is something I misunderstood.

:+1:

stdevAlDen, 2019-05-30:
> I unassigned it from Milestone 1 for now. Let me know if there is something I misunderstood.

@jdknives 
The only thing you need to keep in mind is that in a released device we can make any number of update/fix on firmware, this does not apply for bootloader...
Can you please verify the milestone of [this issue](https://github.com/skycoin/hardware-wallet/issues/239) too?

stdevAlDen, 2019-05-30:
> @stdevAlDen this is not part of Milestone 1. The comments here indicate that this is not critical for release as the user can unplug and reconnect to fix any issue even if this is not the optimal way to go. Unless there is a further reason to work on it, that I am unaware of, do not work on it now. Specficially, I do not understand why you attached the label `critical security`

@jdknives
is important to consider possible scenarios in which a compromised computer can upload a malignant firmware, this is one of the few moments in which the device can become vulnerable, but thinking again on this, the checking firmware signature guarantees the solution of this issue right?