skycoin/hardware-wallet #239: Device not ask for confirmation on upload firmware.

stdevAlDen, 2019-05-30:
**Describe the bug**
After a stuck process of `firmwareUpdate` the device no ask for confirmation on calling `firmwareUpdate`.

**Environment information:**

- OS: GNU/Linux
- Platform: Linux 4.19.0-1-amd64 x86_64
- Firmware Version: https://github.com/skycoin/hardware-wallet/commit/c542b06eae1516896052284f3767abba350fae5d

**Steps to Reproduce**
Steps to reproduce the behavior:

1. Boot the device in bootloader mode.
2. `skycoin-hw-cli firmwareUpdate -f tiny-firmware/skyfirmware.bin`
3. Cancel(<kbd>ctrl</kbd>+<kbd>c</kbd>) the process in the terminal before it get finished.
4. Unplug the device.
5.  Boot the device in bootloader mode.
6. `skycoin-hw-cli firmwareUpdate -f tiny-firmware/skyfirmware.bin`
7. COnfirma the the device no ask you for confirmation about uploading a firmware.

**Actual behavior**
See above steps

**Expected behavior**
Open for discussion.

olemis, 2019-06-02:
> Steps to reproduce the behavior:
> 
>     1. Boot the device in bootloader mode.
> 
>     2. `skycoin-hw-cli firmwareUpdate -f tiny-firmware/skyfirmware.bin`
> 
>     3. Cancel(ctrl+c) the process in the terminal before it get finished.
> 
>     4. Unplug the device.
> 
>     5. Boot the device in bootloader mode.
> 
>     6. `skycoin-hw-cli firmwareUpdate -f tiny-firmware/skyfirmware.bin`
> 
>     7. COnfirma the the device no ask you for confirmation about uploading a firmware.

olemis, 2019-06-02:
This is expected behavior . I'll explain why ...

> 
>     1. Boot the device in bootloader mode.
> 
>     2. `skycoin-hw-cli firmwareUpdate -f tiny-firmware/skyfirmware.bin`
> 
>     3. Cancel(ctrl+c) the process in the terminal before it get finished.
> 

This sequence leads to firmware being erased from device ...

>     4. Unplug the device.
> 
>     5. Boot the device in bootloader mode.
> 

... the proof is that there's no need to press both them wallet buttons at once to enter bootloader mode since that's default behavior (since there's no firmware at all).

>     6. `skycoin-hw-cli firmwareUpdate -f tiny-firmware/skyfirmware.bin`
> 
>     7. COnfirma the the device no ask you for confirmation about uploading a firmware.

no firmware means that there is no need to display message here

closed 2019-06-02 by olemis