skycoin/hardware-wallet #242: When changing the label, confirmation about the passphrase is requested

Senyoret1, 2019-06-02:
**Describe the bug**
When trying to change the label, the device ask for confirmation for the new label, but also ask if the passphrase should be deactivated.

**Environment information:**

- OS: Windows 10
- Platform: X64
- Firmware Version: commit 46ef0d3

**Steps to Reproduce**
1. In the js library repo, go to `/test/test_codes.js`
2. Set the `testApplySettings` const to `false` and make sure that all the other constants for starting tests are set to `false`.
3. A few lines after `testApplySettings` is declared, change the `devApplySettings` call to `devApplySettings(null, 'label', null)`.
4. See what confirmations the device asks for.

Something similar happens with the daemon.

**Actual behavior**
The device asks the user to confirm if the passphrase protection should be deactivated.

**Expected behavior**
The device should not ask for confirmation about the passphrase, as no value about that setting if being provided.

**Additional context**

**Possible implementation**

olemis, 2019-06-02:
>     2\. Set the `testApplySettings` const to `false` and make sure that all the other constants for starting tests are set to `false`.

`false` or `true` ?

olemis, 2019-06-02:
@Did you do anything before , or just followed the steps immediately after either flashing firmware or wiping device ?

Senyoret1, 2019-06-03:
> 
> 
> > ```
> > 2\. Set the `testApplySettings` const to `false` and make sure that all the other constants for starting tests are set to `false`.
> > ```
> 
> `false` or `true` ?

`true`, sorry. I corrected it in the original message.

> 
> 
> @Senyoret1 did you do anything before , or just followed the steps immediately after either flashing firmware or wiping device ?

You can do it just after flashing or after assigning a seed, the result is the same.

olemis, 2019-06-03:
@Senyoret1 could you please share the value you get for `passphraseProtection` as returned by `GetFeatures` message ?

olemis, 2019-06-03:
> @Senyoret1 could you please share the value you get for `passphraseProtection` as returned by `GetFeatures` message ?

Immediately after flashing firmware , and also before and after invoking `ApplySettings`.

Senyoret1, 2019-06-03:
- Just after flashing: `passphraseProtection: false`
- After setting a label with `devApplySettings(null, 'label', null)`: `passphraseProtection: false`
- After setting a seed: `passphraseProtection: false`
- After setting a label with `devApplySettings(true, 'label', null)`: `passphraseProtection: true`

All the steps were made one after another, in order.

stdevAlDen, 2019-06-04:
@Senyoret1 & @olemis this issue does not apply for firmware, fixed in [hw-js](https://github.com/skycoin/hardware-wallet-js/pull/126/files).

closed 2019-06-04 by stdevAlDen

Senyoret1, 2019-06-05:
If the problem is not in the firmware, then it could be also in the go repo or in the daemon repo, as it also happens in the daemon. @therealssj could you please check it? I tried sending `{"label": "skywallet", "use_passphrase": null}` and `{"label": "skywallet"}` to the `apply_settings` endpoint.

stdevAlDen, 2019-06-05:
> label

@Senyoret1 & @therealssj in [hw-go](https://github.com/skycoin/hardware-wallet-go/blob/cba497d953b76490fd52471f9f4f74d0072f7f4d/src/cli/apply_settings.go#L61-L72) is ok...
I no know why in daemon, maybe an outdated version of hw-go or as you say there is an error in hw-go about this.