skycoin/hardware-wallet #243: Setting usePassphase in Recovery message leads to generating wrong addresses

olemis, 2019-06-02:
**Describe the bug**
Setting `usePassphase` in Recovery message leads to generating wrong addresses

**Environment information:**

- OS: Mac OS X
- Platform: Darwin 17.4.0 x86_64
- Firmware Version: 46ef0d346a835263d0c060d074652698954613f7

**Steps to Reproduce**

- Create new wallet e.g. with `skycoin-cli`
```sh
skycoin-cli walletCreate -f $WALLET1.wlt -l $WALLET1
```
- From command output take note of the seed `SEED1` and address `ADDRESS1`
- List wallet addresses and confirm that `ADDRESS1` is the only value in the list.
```sh
skycoin-cli listAddresses $WALLET1.wlt
```
- Ensure device is seedless by [wiping it](https://github.com/skycoin/hardware-wallet-js/tree/master/DOCUMENTATION.md#devWipeDevice). Check that you end up in home screen with `NEEDS SEED!` message at the top.
- [Recover seed](https://github.com/skycoin/hardware-wallet-js/tree/master/DOCUMENTATION.md#devRecoveryDevice) `SEED1` in Skywallet device (`dry-run=false`, `usePassphrase=true`).
- [Generate five addresses using Skywallet](https://github.com/skycoin/hardware-wallet-js/tree/master/DOCUMENTATION.md#devAddressGen)
- Check that address sequence generated by SkyWallet matches the values generated by `skycoin-cli`
```sh
skycoin-cli walletAddAddresses -f $WALLET1.wlt -n 5
```

**Actual behavior**
Skywallet addresses ...

```
26MgaacLTydj2tgcgR1SCAMuDLMvstXwUGG
27SVD1r3Jbb238C2VmHQsdPiTCp7dZN2Ryp
UCo7rQ2vJZTYdRpmebxx44KrVyixnKekuW
2dyk9NuqoWY5L1yLQxqPGn1uoHRmvFmvfbd
2P1mjpJSrbttKt4F6AuX9Vfh48zZYfSyjZf
```

... do not match `skycoin-cli` output

```
Mh9jEhCi23WqSD1SmbJz5PGBArHviHcC2a
eKyAeTqsK3x2Pj1gcSfUzogMRfz85JASzU
2Ene7utsDjhhBuHnBtU15cuUKWpadUkTUDr
7FiHj1ayVBTAfszMDGnV1WCriDMHmT9PAZ
VMsZnKjP6vQQtfUiWRJdCFFSHyoR4H6wSK
```

**Expected behavior**
Provided same seed is recovered on both wallets , Skywallet address sequence must match values generated by `skycoin-cli`

**Additional context**
Discovered while following hw-js pre-release testing procedure skycoin/hardware-wallet-js#112

olemis, 2019-06-03:
Expected behavior . For v1 `passphrase` must always be `false` .

olemis, 2019-06-03:
Nonetheless this should be more visible in dev and user docs .

stdevAlDen, 2019-06-04:
the good news are that the [libskycoin tests](https://github.com/skycoin/hardware-wallet/pull/232) are failing about this, but need to look for more details

jdknives, 2019-06-05:
@olemis if this is expected behavior for v1, why is the Milestone 1 attached? What is to be done about this for v1?

olemis, 2019-06-05:
> What is to be done about this for v1?

Ensure that if `usePassphrase=true` and `passphrase=''` then behavior is the same as `usePassphrase=false` .

PR #296 (2019-07-03): [firmware] fixes #243 - Ignore empty passphrase
Fixes #243 

Changes:	
- Ignore empty passphrase

Does this change need to mentioned in CHANGELOG.md?	
no

Requires testing
yes , added in `fsm_impl.c`
commit 97b946be (2019-07-03): Merge pull request #296 from simelo/olemis_t243_empty_passphrase

[firmware] fixes #243 - Ignore empty passphrase
closed 2019-07-03 by olemis