skycoin/hardware-wallet #244: Label erased after device recovery

olemis, 2019-06-02:
**Describe the bug**
Label erased after device recovery

**Environment information:**

- OS: Mac OS X
- Platform: Darwin 17.4.0 x86_64
- Firmware Version: 46ef0d3

**Steps to Reproduce**

1. Checkout 46ef0d3
2. Run `make clean && make firmware`
3. Flash it onto device e.g. hw-go `firmwareUpdate`
4. Check skycoin/hardware-wallet-js@stdevStark_t112_ci_release
5. Run `node ci-scripts/pre-release.features.js`
6. Run `node ci-scripts/pre-release.recovery.js` and follow recovery procedure for a given wallet seed (exact value does not matter as long as it's a valid mnemonic)
7. Run `node ci-scripts/pre-release.features.js` once again

**Actual behavior**
First time `GetFeatures` is executed this is the result

```
Features {
  vendor: 'Skycoin Foundation',
  deviceId: '35383935065137331B003B00',
  pinProtection: false,
  passphraseProtection: false,
  label: '35383935065137331B003B00',
  initialized: false,
  bootloaderHash:
   Uint8Array [
     176,
     116,
     162,
     71,
     92,
     25,
     181,
     200,
     192,
     109,
     12,
     66,
     210,
     204,
     118,
     222,
     11,
     154,
     16,
     92,
     202,
     210,
     48,
     15,
     50,
     204,
     235,
     97,
     198,
     198,
     245,
     104 ],
  pinCached: false,
  passphraseCached: false,
  needsBackup: false,
  model: '1',
  fwMajor: 1,
  fwMinor: 7,
  fwPatch: 0,
  firmwareFeatures: 0 }
```

Second time it returns the following values . Notice that `label` is missing (i.e. empty).

```
Features {
  vendor: 'Skycoin Foundation',
  deviceId: '35383935065137331B003B00',
  pinProtection: false,
  passphraseProtection: false,
  initialized: true,
  bootloaderHash:
   Uint8Array [
     176,
     116,
     162,
     71,
     92,
     25,
     181,
     200,
     192,
     109,
     12,
     66,
     210,
     204,
     118,
     222,
     11,
     154,
     16,
     92,
     202,
     210,
     48,
     15,
     50,
     204,
     235,
     97,
     198,
     198,
     245,
     104 ],
  pinCached: false,
  passphraseCached: false,
  needsBackup: false,
  model: '1',
  fwMajor: 1,
  fwMinor: 7,
  fwPatch: 0,
  firmwareFeatures: 0 }
```

**Expected behavior**
Device label should not be modified after recovery .

**Additional context**
Detected during hw-js pre-release testing (see skycoin/hardware-wallet-js#112)

Retsediv, 2019-06-06:
Does this bug appear only for hw-js wrapper? Do you have the same functionality for hw-go wrapper (i mean `pre-release.features` and `pre-release.recovery`)?

olemis, 2019-06-19:
> Does this bug appear only for hw-js wrapper?

hw-js , hw-go , and hw-daemon

> Do you have the same functionality for hw-go wrapper (i mean `pre-release.features` and `pre-release.recovery`)?

no

olemis, 2019-06-20:
> > Do you have the same functionality for hw-go wrapper (i mean `pre-release.features` and `pre-release.recovery`)?
> 
> no

Indeed, I reviewed hw-go pre-release once again and some things need to be fixed

Retsediv, 2019-06-24:
@olemis I have just tried to reproduce this feature (on Linux with the same version of firmware and hardware-wallet-js) and it does not appear
Do you have some thought on that?

jdknives, 2019-06-28:
@Retsediv can someone on Mac check whether they can reproduce the issue?

Macrosrider, 2019-07-02:
@olemis The same issue was reproduced on Windows (with the same version of firmware and hardware-wallet-js). So, this is not only Mac OS issue

olemis, 2019-07-04:
> @olemis The same issue was reproduced on Windows (with the same version of firmware and hardware-wallet-js). So, this is not only Mac OS issue

Cool ! Thanks ... @Retsediv any progress ?

Retsediv, 2019-07-04:
> > @olemis The same issue was reproduced on Windows (with the same version of firmware and hardware-wallet-js). So, this is not only Mac OS issue
> 
> Cool ! Thanks ... @Retsediv any progress ?

So far, I could not reproduce it on Linux(and I do not have any other OS). Although, now I have installed a new Linux distro and will recheck everything. Will let you know soon

Retsediv, 2019-07-04:
@olemis unfortunately, I still can not reproduce this bug on Linux

@YuraYelisieiev can you take this issue?

YuraYelisieiev, 2019-07-04:
@Retsediv will finish with my issues and take a look at this one.

olemis, 2019-07-04:
> @Retsediv will finish with my issues and take a look at this one.

@stdevStark will take it fom here .

PR #300 (2019-07-05): [firmware] fixes #244 - Fix label erased after device recovery
Fixes #244 

 Changes:	
- Use current label in `recovery_init` for not remove it if empty

Does this change need to mentioned in CHANGELOG.md?	
no	

Requires testing	
yes , but not possible now .
commit 454a28eb (2019-07-05): Merge pull request #300 from simelo/stdevStark_t244_recovery_keep_label

[firmware] fixes #244 - Fix label erased after device recovery
closed 2019-07-05 by olemis