skycoin/hardware-wallet #247: Update setup instructions

jdknives, 2019-06-04:
**Describe the bug**
brew install protobuf --with-python
gives me: 
Error: invalid option: --with-python

It seems like the setup instructions for the Skywallet are out of date and need to be updated. 

This potentially affects more than just this command.

stdevAlDen, 2019-07-17:
> **Describe the bug**
> brew install protobuf --with-python
> gives me:
> Error: invalid option: --with-python
> 
> It seems like the setup instructions for the Skywallet are out of date and need to be updated.
> 
> This potentially affects more than just this command.

Add install instructions for SDL.

stdevAlDen, 2019-07-17:
> > **Describe the bug**
> > brew install protobuf --with-python
> > gives me:
> > Error: invalid option: --with-python
> > It seems like the setup instructions for the Skywallet are out of date and need to be updated.
> > This potentially affects more than just this command.
> 
> Add install instructions for SDL.

https://github.com/skycoin/hardware-wallet/blob/8a2c42f4a79d0deeb3f7bd776b56c126cf7322c8/ci-scripts/install-osx.sh#L14-L15

jdknives, 2019-09-20:
This needs to be tackled. The current instructions are scattered over two Readmes and the structure could be improved. Will come up with ideas.