skycoin/hardware-wallet #262: test_signature only checks the first 32 bytes of the signature

gz-c, 2019-06-13:
The tests in `test_signature` only check the first 32 bytes of the signature.  If there is a reason for that, it should be explained why, otherwise the full signature should be checked, and it should match the result of the skycoin core libs.

olemis, 2019-06-19:
https://github.com/skycoin/hardware-wallet/blob/develop/skycoin-api/test_skycoin_crypto.c#L377-L378

@gz-c 32 + 32 = 64

olemis, 2019-06-19:
> https://github.com/skycoin/hardware-wallet/blob/aabc77d/skycoin-api/test_skycoin_crypto.c#L377-L378
> 
> @gz-c 32 + 32 = 64

gz-c, 2019-06-20:
I see; this will be fixed in my PR

olemis, 2019-06-20:
Testing for this will be ported from `skycoin/libskycoin` as part of work done for #232

olemis, 2019-06-20:
> it should match the result of the skycoin core libs.

this is thge specific part that will be added in #232

gz-c, 2019-06-20:
The fix was to remove those tests because the funciton it tested was removed.  Then to port those tests to the new function. There isn't a bug, the 32 is the 32 bytes of the R,S points of the signature. The bug was that it wasn't checking the recovery ID.  

This problem is resolve by my PR.

PR #254 (2019-06-20): Cleanup curve parameter access
Fixes #249 
Fixes #263 
Fixes #262 
Fixes #255 
Related to #104 

Changes:	

- Remove erroneous use of `HDNode` as a roundabout way of obtaining the secp256k1 curve parameters
- Remove unnecessary duplicated code in `skycoin_check_signature_tools.{c,h}`
- Remove unused values in `recover_pubkey_from_signed_message`
- Remove decred's curve (we have no intention of supporting decred)
- Add secp256k1.o into the bootloader bin. This could cause the bootloader size to be too large, please check it
- Revert to using the ecdsa_sign_digest function from the ecdsa library [... cut]
closed 2019-06-20 by gz-c