skycoin/hardware-wallet #265: Review firmware_sign.py sign() method

gz-c, 2019-06-16:
**Feature description**

`firmware_sign.py` has a `sign()` method with nonsense modifications. It also call a function `EcdsaSkycoinSign()` which had more nonsense modifications.  Reference the original `firmware_sign.py` code from Trezor and judge whether or not any modification should have been made.

gz-c, 2019-06-21:
#283 improves the python code, after that merges the review can continue

RomanMilishchuk, 2019-06-25:
Most of the changes, that were made are touched with the differences in skycoin and bitcoin. In trezor, they are using ecdsa python lib, in which they use method sign_determinictic, which uses method [sign_digest_deterministic](https://github.com/warner/python-ecdsa/blob/master/src/ecdsa/keys.py#L264). In skycoin realization, all the crypto is doing in C code, which is called from from shared library in python. Digest is computed in [firware_sign.py](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/bootloader/firmware_sign.py#L70) and then uses [skycoin_ecdsa_sign_digest](https://github.com/skycoin/hardware-wallet/blob/develop/skycoin-api/skycoin_crypto.c#L257). But in [ecdsa_sign_digest](https://github.com/skycoin/hardware-wallet/blob/develop/skycoin-api/tools/ecdsa.c#L702) k is chosen randomly, which makes such signing non-deterministic. 

So to sum up, Trezor is finding deterministic signature, but Skycoin non-deterministic. As in hardware-wallet code related to crypto is not duplicated in Python it is generally much better, but the question how usage of non-deterministic signatures instead of determinstic influece others parts.

gz-c, 2019-06-25:
RFC6979 support (deterministic signatures) were removed by the original developer on this project and I haven't felt that it was necessary to put them back in, since Skycoin doesn't implement RFC6979 in its own signing libraries.  Note that a deterministic signature is still a valid Skycoin signature.  The "deterministic" describes how the signature is generated, but the nature of the signature is unchanged.

In this review, I would like people to look for bugs as well.  For example, misaligned memory reads, since I have changed the memory layout in the bootloader.

RomanMilishchuk, 2019-06-25:
Size of meta is now 257 bytes and I think for aligning it is better to make it 256, as there are reserved bytes. Also [RESERVED_LEN](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/bootloader/firmware_sign.py#L70) is hardcoded for 3 slots and instead should be rewritten to calculate depending on number of slots. Also, it should be better to add constant for size of meta (would be useful [here](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/bootloader/firmware_sign.py#L70) ).

gz-c, 2019-06-25:
Ok, If you can align it. Note that signatures are 65 bytes each, not 64 bytes each, so that forces the alignment to be off.  I manually adjusted the offsets to reduce the reserved space, and give more room for the signatures.

RomanMilishchuk, 2019-06-25:
If it is frozen into bootloader, then you can change a number of slots before writing bootloader into skywallet. Maybe it won't be used, but it is definitely better, than magical constant 49. 

As I understood after meta info is going code and this part can be aligned. One byte can be easily taken from reserved memory without consequences.

In which C file there are such offsets?

olemis, 2019-06-25:
>  a dynamic slot size may not be very helpful

👍

olemis, 2019-06-25:
> If it is frozen into bootloader, then you can change a number of slots before writing bootloader into skywallet. Maybe it won't be used, but it is definitely better, than magical constant 49.

I'm not quite sure about this change ...

gz-c, 2019-06-26:
@RomanMilishchuk these offsets are in https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/memory.h

olemis, 2019-06-26:
> Most of the changes, that were made are touched with the differences in skycoin and bitcoin.

Why not just using [PySkycoin](https://github.com/skycoin/pyskycoin) for invoking from Pythob the very same crypto API's present in Skycoin cipher library ?

RomanMilishchuk, 2019-06-26:
This question was needed not now, but a few months ago. However, PySkycoin and skycoin-api are both wrappers using other languages, so I don't see much differences between them.

gz-c, 2019-06-26:
The python wrapper here simply wraps the underlying C library used by the hardware wallet.  We do not need to import a second library.  In fact, there is no reason to use skycoin signatures for the bootloader at all and I don't know why it was changed to use them in the first place.

jdknives, 2019-06-27:
@gz-c So it should be changed back to the Trezor implementation?

olemis, 2019-07-03:
For the record ...

> @olemis This question was needed not now, but a few months ago. 

This issue was opened 17 days ago as a result of the crypto review . I do not see why we should have asked this question months ago given the fact that we are not in charge of the crypto review . More important than the exact moment in time for reporting this is the fact that this needs to be fixed before releasing v1 ; so @RomanMilishchuk , since I see only few issues assigned to you , would you mind completing this one ? I already assigned it to you .

> However, PySkycoin and skycoin-api are both wrappers using other languages, so I don't see much differences between them.

The fact that you see no difference between them makes me wonder whether you understood anything at all in the first place ; but the explanation is beyond the scope of this issue .

olemis, 2019-07-03:
> so @RomanMilishchuk , since I see only few issues assigned to you , would you mind completing this one ? I already assigned it to you .

if this is just about reverting code back to Trezon implementation , it should be an easy task

gz-c, 2019-07-03:
whichever is easier and less likely to have a bug: reverting or verifying the changes

RomanMilishchuk, 2019-07-04:
Trezor checks genuineness of the firmware by next 3 properties:
- Size of firmware
- Hash of firmware
- Signatures

Skywallet now has only check using signatures. As for me, the crypto part is working well (it just signs firmware with a public key and then while checking signatures recover a public key from signature and checks whether this public key is in the list above)

The problems, that I see, that if someone wants to load malicious firmware, then he can do it easily, use malicious firmware and append to the front meta from binaries releases, that are on GitHub. As a result, the bootloader will check signatures, they will be the same, so bootloader will find malicious firmware as proper.

To summarize, for now, it should be reverted to Trezor or added checks for the size of firmware and hashes of the firmware.

jdknives, 2019-07-04:
Reverting to Trezor sounds like the better option. @RomanMilishchuk

RomanMilishchuk, 2019-07-04:
But then can be a problem with memory, because @gz-c mentioned, that he changed the offsets.

stdevAlDen, 2019-07-04:
> Reverting to Trezor sounds like the better option. @RomanMilishchuk

@jdknives I no think so, making that will break [this feature](https://github.com/skycoin/hardware-wallet/pull/281).

stdevAlDen, 2019-07-04:
> Trezor checks genuineness of the firmware by next 3 properties:
> 
> * Size of firmware

This is currently checked by the tooling.

> * Hash of firmware
> * Signatures

 Looks redundant for me, do you think that checking the signature of the hash is not enough?

> Skywallet now has only check using signatures. As for me, the crypto part is working well (it just signs firmware with a public key and then while checking signatures recover a public key from [...]

RomanMilishchuk, 2019-07-09:
@stdevAlDen Bootloader, for now, checks only signatures and I have written why it is bad and how can you use this, so why do you ask again why it is not enough without any arguments?

stdevAlDen, 2019-07-09:
> @stdevAlDen Bootloader, for now, checks only signatures and I have written why it is bad and how can you use this, so why do you ask again why it is not enough without any arguments?

@RomanMilishchuk my simple argument is that if you change the firmware, the hash will change and the signature check will fail. Can you please share where you put the "I have written why it is bad and how can you use this"?

RomanMilishchuk, 2019-07-09:
Here I have written, why it is bad and how to use it.
> The problems, that I see, that if someone wants to load malicious firmware, then he can do it easily, use malicious firmware and append to the front meta from binaries releases, that are on GitHub. [...]

For checking signatures are used function [skycoin_ecdsa_verify_digest_recover](https://github.com/skycoin/hardware-wallet/blob/454a28ebbfb8bc3d77f835b88a3044c04d974bd0/tiny-firmware/bootloader/signatures.c#L100), which uses ecdsa_verify_digest_recover and what it do is just recovering pubkey from signature and then [here](https://github.com/skycoin/hardware-wallet/blob/454a28ebbfb8bc3d77f835b88a3044c04d974bd0/tiny-firmware/bootloader/signatures.c#L106) this public keys are compared.

To sum up, if you change the firmware, the hash will change and the signature check will NOT fail, so it is why you have to check hashes indepentenly.

Another way, could be to generate signature for firmware and check generated signature and one that is in firmware, but it isn't option for skycoin, because, as @gz-c mentioned:
> RFC6979 support (deterministic signatures) were removed by the original developer on this project and I haven't felt that it was necessary to put them back in, since Skycoin doesn't implement RFC6979 in its own signing libraries. 
So if these signatures are not same it doesn't mean, that it is malicious firmware.

stdevAlDen, 2019-07-09:
> Here I have written, why it is bad and how to use it.
> 
> > The problems, that I see, that if someone wants to load malicious firmware, then he can do it easily, use malicious firmware and append to the front meta from binaries releases, that are on GitHub. As a result, the bootloader will check signatures, they will be the same, so bootloader will find malicious firmware as proper.
> 
> For checking signatures are used function [skycoin_ecdsa_verify_digest_recover](https://github.com/skycoin/hardware-wallet/blob/454a28ebbfb8bc3d77f835b88a3044c04d974bd0/tiny-firmware/bootloader/signatures.c#L100), which uses ecdsa_verify_digest_recover and what it do is just recovering pubkey from signature and then like [here](https://github.com/skycoin/hardware-wallet/blob/454a28ebbfb8bc3d77f835b88a3044c04d974bd0/tiny-firmware/bootloader/signatures.c#L106) these public keys are compared.
> 
> To sum up, if you change the firmware, the hash will change and the signature check will NOT fail, so it is why you have to check hashes indepentenly.
> 
> Another way, could be to generate signature for firmware and check generated signature and one that is in firmware, but it isn't option for skycoin, because, as @gz-c mentioned:
> 
> > RFC6979 support (deterministic signatures) were removed by the original developer on this project and I haven't felt that it was necessary to put them back in, since Skycoin doesn't implement RFC6979 in its own signing libraries.
> > So if these signatures are not same it doesn't mean, that it is malicious firmware.

@RomanMilishchuk great, thank you, now I see the problem, if I can come with a solution proposal I'll come back to this.

gz-c, 2019-07-10:
- This code is fully derived from https://github.com/trezor/trezor-mcu/tree/master/bootloader
- The code was modified by someone who does not work for us anymore
- These modifications had some mistakes, which I fixed. But, there can be more mistakes
- Assume the original code from trezor was not wrong
- Trezor did not always have deterministic signatures, so this system should work without them.  In other words, signature determinism is irrelevant
- Verifying a signature requires both a hash of the firmware and a known public key to compare against

gz-c, 2019-07-10:
>To sum up, if you change the firmware, the hash will change and the signature check will NOT fail, so it is why you have to check hashes indepentenly.

If the hash changes, then `skycoin_ecdsa_verify_digest_recover` can either fail completely (return value nonzero), or return a pubkey that does not match the configured pubkeys.  Most of the time it will return a valid pubkey. It is required that the caller compare the recovered pubkey to the list of hardcoded pubkeys, to make sure that the pubkey matches.

RomanMilishchuk, 2019-07-10:
@gz-c then looks, like everything is ok.

One point, that while computing the hash of firmware, then the size of firmware is just used from the [meta](https://github.com/skycoin/hardware-wallet/blob/454a28ebbfb8bc3d77f835b88a3044c04d974bd0/tiny-firmware/bootloader/signatures.c#L72), so there is an option to append new code, but I think that there is no problem, as this code would never be called.

jdknives, 2019-07-12:
@RomanMilishchuk @gz-c @olemis Will close this as there are no more commenrs as far as I can see and nothing needs to be fixed.

closed 2019-07-12 by jdknives

olemis, 2019-07-12:
@jdknives is there an open PR for this ?