skycoin/hardware-wallet #266: Firmware panic

olemis, 2019-06-17:
**Feature description**

Add ability to freeze firmware execution

**Is your feature request related to a problem? Please describe.**
If critical crypto conditions detected then device should be locked to prevent access to coins and preserve funds in wallet.

**Describe the solution you'd like**
Implement a function that relies on built-in device traps to halt device execution .

**Describe alternatives you've considered**
Setting a flag in order to exit main loop gracefully , might be hijacked (i.e. flag value reverted after failure detected but prior to loop conditional).

**Additional context**
Skycoin core panics if some (serious) crypto errors are detected at run time . This is to protect coins . Firmware does not support doing so .

**Possible implementation**
TBD

gz-c, 2019-06-17:
I asked if there was `abort()` function, and if there isn't, then its not a problem. We can use `return`.

olemis, 2019-06-17:
@gz-c there is a way (already implemented) , just looking for the best to trigger it by hardware

olemis, 2019-06-22:
firmware panic is safer since after entering a hardware trap, this aborts any further processing

jdknives, 2019-07-17:
@olemis you removed the Milestone 1 label 18 days ago but it is still attached to it via the Milestone feature. Do we need to fix this for v1?

olemis, 2019-07-17:
this is not v1 feature