skycoin/hardware-wallet #269: Configure Travis to automatically upload new binaries

jdknives, 2019-06-18:
**Feature description**

We will need to configure Travis to automatically upload new firmware binaries to the Skycoin domain. This way, we do not have to manually do this with new releases.

olemis, 2019-06-18:
Travis already deploys to AWS bucket

olemis, 2019-06-18:
I'm reopening this issue because I noticed `deploy` was triggered for `master` branch , and this should be happening for tags .

Senyoret1, 2019-06-19:
I think the issue is about automatically uploading the compiled firmware to https://downloads.skycoin.net/skywallet/skyfirmware_{firmware-version}.bin, so the desktop wallet can find it. However, for this to work in a totally automatic way, when uploading the firmware the https://version.skycoin.net/skywallet/version.txt file would have to be updated to contain the version number of the recently uploaded firmware. This is because the desktop wallet works like this:

- For knowing if the hardware wallet firmware should be updated, it checks the version number stored in https://version.skycoin.net/skywallet/version.txt and compares it to the one in the currently connected hw wallet.

- For updating the firmware, it checks the version number stored in https://version.skycoin.net/skywallet/version.txt, saves it in a variable (lets call it `v_`) and then dowloads the firmware from `'https://downloads.skycoin.net/skywallet/skyfirmware_' + v_ + '.bin'`

Also, it is important to keep in mind that those URLs are the ones that will be used in production, so after launching v1 only production ready firmwares should be uploaded there. If we want to have dev preview versions for testing, we will have to use another URLs.

stdevAlDen, 2019-06-19:
> desktop

@Senyoret1 | @jdknives there is any project currently uploading content to https://downloads.skycoin.net to follow the steps on how should I make this? If not, can you please give me any hint?

Senyoret1, 2019-06-19:
I don't know about other projects uploading to that server. I think @gz-c knows how the server is configured

stdevAlDen, 2019-06-19:
> I think the issue is about automatically uploading the compiled firmware to [...]

@Senyoret1 the skycoin version scheme have a `v` as prefix, this is how the version.txt value should look like or do you need me to remove the  `v`?

contain the version number of the recently uploaded firmware. This is because the desktop wallet works like this:
> * For knowing if the hardware wallet firmware should be updated, it checks the version number stored in https://version.skycoin.net/skywallet/version.txt and compares it to the one in the currently [...]

Senyoret1, 2019-06-19:
@stdevAlDen You can put the v or skip it, the desktop wallet code works correctly in both cases. However, if you add the v at the beginning of the version number, you must also add it to the name of the firmware file. For example, if the file with the lastest version contains `v1.0.1`, the firmware file should be named `skyfirmware_v1.0.1.bin`.

olemis, 2019-06-19:
@Senyoret1 at first sight Travis deployment is correct but we better try this .

PR #276 (2019-06-19): [WIP] deploy on tags ref #269
Fixes #269

Changes:	
 - Deploy in tags

Does this change need to mentioned in CHANGELOG.md?	
no

Requires testing	
no
commit 3c98a266 (2019-06-19): Merge pull request #276 from simelo/stdevAlDen_t269_deploy_on_tags

Deploy on tags fixes #269
closed 2019-06-19 by olemis