skycoin/hardware-wallet #284: Compilation fails due to the large bootloader size

Retsediv, 2019-06-21:
**Describe the bug**
After compiling(under Linux) the bootloader and firmware, the command `make full-firmware` fails due to the large bootloader size(it should be less than 32768 bytes).

The error after `make full-firmware`:
```
Traceback (most recent call last):
  File "./firmware_align.py", line 8, in <module>
    raise Exception('bootloader has to be smaller than 32768 bytes')
Exception: bootloader has to be smaller than 32768 bytes
Makefile:64: recipe for target 'align' failed
make[1]: *** [align] Error 1
```

From our initial research, this bug does not appear on MacOS, only on Linux.

**Environment information:**

- OS: Linux
- Platform: Linux 4.15.0-20-generic x86_64
- Firmware Version: latest `develop` commit

**Steps to Reproduce**

1. Use Linux
2. Checkout the `develop` branch
3. Clean everything `make clean && make clean-lib`
4. Compile bootloader `make build-deps && make bootloader`
5. Compile firmware `make firmware-deps && make firmware`
6. Combine them together `make full-firmware`

**Actual behaviour**
An error occurs

**Expected behaviour**
No error occurs and everything successfully compiles

**Possible implementation**
First of all, need to check the actual size of bootloader compiled on Linux and MacOS. Then, try to understand where is the "overflow" happens. Probably, it's due to some external libraries, which are different for Linux and MacOS

Retsediv, 2019-06-28:
After better investigation(with @Macrosrider), we understood that it happens because after the execution of (3) and (4) commands(from **Steps to Reproduce**) the command `make full-firmware` does not case of previously compiled bootloader and firmware and recompile everything. During that, it (do not know why) append new data to the bootloader file(so it becomes twice big) and the error( `bootloader has to be smaller than 32768 bytes` ) appears.

The possible quick solution is in the following PR #293. Although, there is a problem in this PR with signing and etc, which has to be solved

Another solution could be to rewrite Makefile, so all targets will correctly understand which dependencies were already compiled and which not. So, it will give us much better compilation time(my guess is at least 2-3 times faster) and fix this error in a much better way.  

So, my proposition is to use #293 for now (after some changes), but refactor Makefile after v1 release
@jdknives @olemis your thoughts?

stdevAlDen, 2019-06-28:
> After better investigation(with @Macrosrider), we understood that it happens because after the execution of (3) and (4) commands(from **Steps to Reproduce**) the command `make full-firmware` does [...]

I have updated the #293, let me know if this now works

jdknives, 2019-07-04:
@Retsediv @stdevAlDen closed  by #293

closed 2019-07-04 by jdknives