skycoin/hardware-wallet #286: Verify set_up_rdp_level is called with the right flags

gz-c, 2019-06-22:
**Feature description**

`set_up_rdp_level()` from `memory.c` is supposed to set the RDP level based upon some flags.  Describe how and where these flags are set, and verify that the correct RDP level is set for production builds.

stdevAlDen, 2019-06-26:
> set_up_rdp_level

@gz-c What do you means with "...verify that... " in this issue? Add an entry for [the prerelease is enough](https://github.com/skycoin/hardware-wallet#pre-release-testing)?

stdevAlDen, 2019-06-28:
@gz-c `set_up_rdp_level` not do anything in the physical device, just read the rdp level [from the device](https://github.com/skycoin/hardware-wallet/blob/12968889ca8d5e96201e2defeb92e2c137911f73/tiny-firmware/memory.h#L129-L135) into a variable, `rdp_level`, the function name have th prefix `set up_`. This is important because after the mpu get enabled I can not read this memory region and it's required to get this info from the [features command](https://github.com/skycoin/hardware-wallet/blob/12968889ca8d5e96201e2defeb92e2c137911f73/tiny-firmware/firmware/fsm_impl.c#L289-L342). I you are interested into know how this value is set for the physical dev just look at [this](https://github.com/skycoin/hardware-wallet/blob/12968889ca8d5e96201e2defeb92e2c137911f73/tiny-firmware/memory.c#L39-L50).

Let me know if you need any thing else for this issue, maybe I'm missing some thing.

PD: I can not check that this is working with my device.

olemis, 2019-06-28:
@stdevAlDen so it does not actually write values on RDP-related registers , right ? If so , I'd better rename that function to be copy_rdp_level or something alike

olemis, 2019-07-04:
> PD: I can not check that this is working with my device.

@jdknives already confirmed this was working as expected

Retsediv, 2019-07-04:
After the code review, it looks like everything is set correctly, but later will check it on the physical device by trying to access the protected memory region.

@jdknives as I understood, you had skywallet with enabled RDP(like 2nd level) and could not access flash memory.

olemis, 2019-07-11:
@stdevstark pls take a look into this and finish asap . Thnx in advance .

stdevStark, 2019-07-11:
ok

stdevStark, 2019-07-11:
@olemis see #305 . Can you test with device? I have no device with RDP .

PR #305 (2019-07-17): [firmware] [bootloader] fixes #286 - Change names of RDP level
Fixes #286 

 Changes:	
- Implement `memory_rdp_level()`
- Remove `extern rdp_level` and `set_up_rdp_level()`

Does this change need to mentioned in CHANGELOG.md?	
no	

 Requires testing	
yes | no	

Impossible to test if not have one device
commit 8cc2e992 (2019-07-17): Merge pull request #305 from simelo/stdevStark_t286_setup_rdp

[firmware] [bootloader] fixes #286 - Change names of RDP level
closed 2019-07-17 by olemis