skycoin/hardware-wallet #288: Review entropy.c and use of entropy

gz-c, 2019-06-22:
**Feature description**

- Review entropy.c for correctness
- Review the use of random numbers in the codebase, ensuring that the entropy is actually used

YuraYelisieiev, 2019-06-26:
1. In [gpio_noise.c](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/firmware/gpio_noise.c#L29) salt is implemented strange. In this [line](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/firmware/gpio_noise.c#L51) you are XORING right side of the returning value with the salt, but the sault is accumulating every time, so it is predictable and we can guess the whole value. Also, the question is how uint16_t behaves after the overflow.

2. When we are mixing entropy from two different sources [here](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/firmware/entropy.c#L134)
we are using SHA256 as an entropy extractor(accumulating the entropy sources). We are using sha256 multiple times in a row, is there any sense of doing it. I think we need to research more about the entropy mixing and choose between different extractors.

RomanMilishchuk, 2019-06-26:
In [rng.c](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/rng.c) random32 works wrong, because no 2 equal numbers can go in sequence. Actually, this file even isn't needed because the same function [rng_get_random_blocking](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/vendor/libopencm3/include/libopencm3/stm32/common/rng_common_v1.h#L79) (that works properly) is implemented by ARM in libopencm3 library (which hardware-wallet included). Also even TRNG with random32 is written in rng.c, there is no declaration of it in rng.h and is included rand.h, which has random32 function that will don't work for STM, because of the usage of /dev/urandom file.

In [random_salted_buffer](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/firmware/entropy.c#L173) random_buffer is using withot if EMULATOR directive, so will have bugs, when using this function on hw, as random_buffer has proper implementation only for PC in skycoin-crypto.

What is sense of making here uint16_t from uint8_t in [gpio_noise.c](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/firmware/gpio_noise.c#L51) with using salt as iterative variable (which makes such random very bad) and then this uint16_t is truncated to uint8_t [here](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/firmware/entropy.c#L116).

gz-c, 2019-06-27:
For the "salt": the purpose is to break a certain kind of RNG attack, where the RNG has a secret hidden pattern, which the attacker knows how to break.

The salt is not intended to be a source of entropy.

XORing two different values does not reduce the overall entropy, even if one value has no entropy.

The salt mixer is based upon the mixer in here: https://github.com/skycoin/skycoin/blob/develop/src/cipher/secp256k1-go/secp256_rand.go
which XORs an sha256 hash of salt values with the system's entropy stream.  It should achieve the same result.

gz-c, 2019-06-27:
The problem I have is that i see `random32()` is NOT using the entropy pool mixer, and @olemis's PR to fix it, does not implement it the way I want.  

I am also uncertain that the correct rand() functions are being called, it was too difficult to read.  I could not easily verify that all calls to rand() by the libraries (e.g., `random32() in `generate_k_random()` or the call in `bip39.c`) were using the salted entropy.

YuraYelisieiev, 2019-06-28:
And what about salt in GPIO noise?

YuraYelisieiev, 2019-06-28:
If we always will use the same pin and port [here](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/firmware/entropy.c#L115) we can reduce a semantic load in our function [read_gpio_noise](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/firmware/gpio_noise.c#L21) and setup only one PIN on the PORT, or in the case if we will change PORT and PIN setup all of them before. I can be wrong, but it`s a bad practice to setup them during a function call.

RomanMilishchuk, 2019-06-28:
For the LR register, there is no much sense of using it, as entropy salt, because mostly it will be used in storage_init func, so the address of returning from reset_entropy_mix_256 will be near always the same. For the different devices, this address won`t be the same only in situation, when they have different firmware versions. 

For now, there is no much sense of talking about the correctness of entropy.c, before @olemis's PR will be finished and merged. (Even in this PR, TRNG of STM32 is not used)

stdevAlDen, 2019-06-29:
@YuraYelisieiev please be careful with source code links branches can move fordwar

> 1. In [gpio_noise.c](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/firmware/gpio_noise.c#L29) salt is implemented strange. In this [...]

@YuraYelisieiev Can you predict the value of [gpio_port_read](https://github.com/skycoin/hardware-wallet/blob/12968889ca8d5e96201e2defeb92e2c137911f73/tiny-firmware/firmware/gpio_noise.c#L49)?

[this](https://github.com/skycoin/hardware-wallet/blob/12968889ca8d5e96201e2defeb92e2c137911f73/tiny-firmware/firmware/gpio_noise.c#L51) is just to get an amplification of the readed value.

and we can guess the whole value. Also, the question is how uint16_t behaves after the overflow.
> 2. When we are mixing entropy from two different sources [here](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/firmware/entropy.c#L134)
>    we are using SHA256 as an entropy [...]

stdevAlDen, 2019-06-29:
> In [rng.c](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/rng.c) random32 works wrong, because no 2 equal numbers can go in sequence. Actually, this file even isn't needed because the same function [rng_get_random_blocking](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/vendor/libopencm3/include/libopencm3/stm32/common/rng_common_v1.h#L79) (that works properly) is implemented by ARM in libopencm3 library (which hardware-wallet included). Also even TRNG with random32 is written in rng.c, there is no declaration of it in rng.h instead and is included rand.h from skycoin-crypto, which has random32 function that will don't work for STM, because of the usage of /dev/urandom

@RomanMilishchuk I thing that this is not possible because require POSIX, can you see the compile time condition for EMULATOR or device, there are two implementation.

 file.
> 
> In [random_salted_buffer](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/firmware/entropy.c#L173) random_buffer is using withot if EMULATOR directive, so will have bugs, when using this function on hw, as random_buffer has proper implementation only for PC in skycoin-crypto.
> 
> What is sense of making uint16_t from uint8_t in [gpio_noise.c](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/firmware/gpio_noise.c#L51) with using salt as iterative variable (which makes such random very bad) and then this uint16_t is truncated to uint8_t [here](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/firmware/entropy.c#L116).

stdevAlDen, 2019-06-29:
@YuraYelisieiev & @RomanMilishchuk please is important to get [a permanent link](https://help.github.com/en/articles/getting-permanent-links-to-files#press-kbdykbd-to-permalink-to-a-file-in-a-specific-commit) because branches can move forward and link can get broken.

YuraYelisieiev, 2019-06-29:
> @YuraYelisieiev please be careful with source code links branches can move fordwar
> 
> > 1. In [gpio_noise.c](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/firmware/gpio_noise.c#L29) salt is implemented strange. In this [line](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/firmware/gpio_noise.c#L51) you are XORING right side of the returning value with the salt, but the sault is accumulating every time, so it is predictable
> 
> @YuraYelisieiev Can you predict the value of [gpio_port_read](https://github.com/skycoin/hardware-wallet/blob/12968889ca8d5e96201e2defeb92e2c137911f73/tiny-firmware/firmware/gpio_noise.c#L49)?
> 
> [this](https://github.com/skycoin/hardware-wallet/blob/12968889ca8d5e96201e2defeb92e2c137911f73/tiny-firmware/firmware/gpio_noise.c#L51) is just to get an amplification of the readed value.
> 
> and we can guess the whole value. Also, the question is how uint16_t behaves after the overflow.
> 
> > 1. When we are mixing entropy from two different sources [here](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/firmware/entropy.c#L134)
> >    we are using SHA256 as an entropy extractor(accumulating the entropy sources). We are using sha256 multiple times in a row, is there any sense of doing it. I think we need to research more about the entropy mixing and choose between different extractors.

@stdevAlDen Yeah you're right we can't predict an outcome from the port read. Did not think about tan amplification.

YuraYelisieiev, 2019-06-29:
@stdevAlDen But where is a sense in amplification because you are then just cutting off the part which you added to a return number [here](https://github.com/skycoin/hardware-wallet/blob/12968889ca8d5e96201e2defeb92e2c137911f73/tiny-firmware/firmware/entropy.c#L115)?

stdevIvr, 2019-07-01:
@YuraYelisieiev There are some important things that you need to know to fully understand what's happening here.
1. The reading from a pin (analog or digital) is always a voltage level, and voltage levels are differences of pottential, so they needs to be referenced to ground.
2. Since we are reading without a fixed reference to ground, we are getting just noise and even when you move the device the noise affect that reading and is totally unpredictable.
3. The true random value readed, is very small, and needs to be amplified and is also mixed with itself before get mixed with the actual entropy.

YuraYelisieiev, 2019-07-01:
@stdevIvr Thanks for the explanation, It was basic knowledge to start the review, which I got. But then you are cutting it to uint8 and not using the amplified part. So why you need to amplify it then ?

stdevIvr, 2019-07-01:
@YuraYelisieiev I don't know what do you say that is not used... 

```cpp
entropy_mix_256((uint8_t*)&salt_gpio, sizeof(salt_gpio), NULL);
```

olemis, 2019-07-02:
> The problem I have is that i see `random32()` is NOT using the entropy pool mixer, and @olemis's PR to fix it, does not implement it the way I want.

this should be implemented as expected now .

olemis, 2019-07-02:
> I am also uncertain that the correct rand() functions are being called, it was too difficult to read. I could not easily verify that all calls to rand() by the libraries (e.g., `random32() in `generate_k_random()`or the call in`bip39.c`) were using the salted entropy.

all instances of `random32()` salt random numbers using entropy pool .

YuraYelisieiev, 2019-07-03:
> @YuraYelisieiev I don't know what do you say that is not used...
> 
> ```c++
> entropy_mix_256((uint8_t*)&salt_gpio, sizeof(salt_gpio), NULL);
> ```
@stdevIvr Amplification increase the size to uint16 but then you are casting it to uint8, so the amplified part disappears. Because the bit row can`t fit into this type, so it cuts and the first uint16 part leaves.

isaacvr, 2019-07-08:
> @stdevIvr Amplification increase the size to uint16 but then you are casting it to uint8, so the amplified part disappears. Because the bit row can`t fit into this type, so it cuts and the first uint16 part leaves.

No, it doesn't disappear. That line take the `uint16`, get the location and then use the two `uint8` parts of it. It's a basic pointer operation.

YuraYelisieiev, 2019-07-08:
> No, it doesn't disappear. That line take the `uint16`, get the location and then use the two `uint8` parts of it. It's a basic pointer operation.
That`s what I am talking about. If we use only uint8 part, why we need to amplify it?

isaacvr, 2019-07-08:
@YuraYelisieiev This is something that you should already know:

`uint16` =>    xxxxxxxx      xxxxxxxx
`uint8` =>    xxxxxxxx

If you take a pointer of the first one, and cast it to `uint8`, and then you use the `sizeof` to get the bytes of that first one data (2 bytes), you'll be able to iterate through all the data, there is not missing information here. Like I told you, that's basic pointer operations. That's all for me.

YuraYelisieiev, 2019-07-08:
> @YuraYelisieiev This is something that you should already know:
>
> `uint16` => xxxxxxxx xxxxxxxx
> `uint8` => xxxxxxxx
>
> If you take a pointer of the first one, and cast it to `uint8`, and then you use [...]

@isaacvr I know it thanks.

PR #292 (2019-07-19): fixes #287 #288 - Salt random values used for ECDSA nonces
Fixes #287
Fixes #288 

Changes:	
- Set RNG to `random_salted_buffer` during firmware compilation

Does this change need to mentioned in CHANGELOG.md?	
no

Requires testing	
no	

Observations

Bootloader size might change .
closed 2019-07-19 by gz-c