skycoin/hardware-wallet #291: The device stops working correctly after trying to generate addresses

Senyoret1, 2019-06-23:
**Describe the bug**
When trying to get adresses from the device, it gets stuck and stops responding correctly to the next requests. The problem is fixed by disconnecting and reconnecting the device.

**Environment information:**

- OS: Windows 10
- Platform: X64
- Firmware Version: #1296888

**Steps to Reproduce**
With the hw wallet daemon: call the `generate_addresses`  endpoint requesting the first address of the device (in fact it fails when requesting other adresses as well).

With the hw JS library: call the `devAddressGen` function requesting the first address of the device (it also fails when requesting other adresses).

**Actual behavior**
The device does not respond, or does it after a lot of time (I’m not sure about this last option).

**Expected behavior**
The device should respond, just as it does when other requests are made.

**Additional context**
Tested with the lastest firmware in the develop branch and having run `git pull --recurse-submodules`

**Possible implementation**

olemis, 2019-06-23:
I built firmware at 12968889ca8d5e96201e2defeb92e2c137911f73 and tested with skycoin/hardware-wallet-go@bcda77814889b01f8fddaa637d7eff57205adbd2 .

```
$ time go run cmd/cli/cli.go addressGen --addressN 1 --startIndex 0 --deviceType USB

real	0m3.031s
user	0m1.098s
sys	0m1.185s
```

@Senyoret1 how many addresses are you creating ?

Senyoret1, 2019-06-23:
Just the first one

olemis, 2019-06-23:
> I built firmware at [1296888](https://github.com/skycoin/hardware-wallet/commit/12968889ca8d5e96201e2defeb92e2c137911f73) and tested with [skycoin/hardware-wallet-go@bcda778](https://github.com/skycoin/hardware-wallet-go/commit/bcda77814889b01f8fddaa637d7eff57205adbd2) .

... same as above but generating 99 addresses

```
$ time go run cmd/cli/cli.go addressGen --addressN 99 --startIndex 0 --deviceType USB
...

real	0m41.433s
user	0m1.219s
sys	0m1.201s
```

olemis, 2019-06-23:
> Just the first one

what are the timings you get ?

Senyoret1, 2019-06-23:
I have not tried with hardware-wallet-go, just the JS lib and the daemon. There is not response at all. Let me try again with hardware-wallet-go

olemis, 2019-06-23:
> I have not tried with hardware-wallet-go, just the JS lib and the daemon. There is not response at all. Let me try again with hardware-wallet-go

Cool . Per the results about the issue does not seem to be related to firmware . Would be nice to narrow the root causes and move it to the right repository .

p.s. @gz-c I'm considering the timings listed above are within expected boundaries .

olemis, 2019-06-23:
> > I have not tried with hardware-wallet-go, just the JS lib and the daemon. There is not response at all. Let me try again with hardware-wallet-go
> 
> Cool . Per the results about the issue does not seem to be related to firmware . Would be nice to narrow the root causes and move it to the right repository .
> 
> p.s. @gz-c I'm considering the timings listed above are within tolerance boundaries .

Senyoret1, 2019-06-24:
I just tested with the lastest version of hardware-wallet-go. I ran `go run cmd/cli/cli.go addressGen --addressN 1 --startIndex 0 --deviceType USB` and there was no response. `go run cmd/cli/cli.go features --deviceType USB` worked well. Just in case, I compiled the firmware again with `make full-firmware`, flashed the full firmware with a ST-Link and tested again in Windows and Linux, but the result was the same.

Not sure why it could be giving different results, May it be the submodule? Do the hw wallet you used has a seed?

olemis, 2019-06-24:
> Do the hw wallet you used has a seed?

of course , else if would have returned an error message instead of address list

olemis, 2019-06-24:
> I compiled the firmware again with `make full-firmware`, flashed the full firmware with a ST-Link and tested again in Windows and Linux, but the result was the same.

JFTR , I tested on OS X

olemis, 2019-06-24:
> > I compiled the firmware again with `make full-firmware`, flashed the full firmware with a ST-Link and tested again in Windows and Linux, but the result was the same.
> 
> JFTR , I tested on OS X

@stdevAlDen can you reproduce this on GNU/Linux ?

stdevAlDen, 2019-06-24:
> > > I compiled the firmware again with `make full-firmware`, flashed the full firmware with a ST-Link and tested again in Windows and Linux, but the result was the same.
> > 
> > 
> > JFTR , I tested on OS X
> 
> @stdevAlDen can you reproduce this on GNU/Linux ?

@olemis nop, just works, let me try with the daemon.
@Senyoret1 what daemon version are you trying with?

olemis, 2019-06-24:
@stdevAlDen ... please do so . It does not look like something going wrong with firmware though .

Senyoret1, 2019-06-24:
Lastest from master branch

olemis, 2019-06-24:
> Lastest from master branch

@Senyoret1 we can't reproduce this issue our side . Could you maybe go back in history up to the point when this seems to be starting to fail ?

stdevAlDen, 2019-06-24:
> Lastest from master branch

@olemis  & @Senyoret1  unfortunately I was no able to reproduce the problem, I was able to get an address from the web app with the device connected.
Any other hint to make this happen?
Linux 4.19.0-1-amd64 x86_64

PR #293 (2019-07-03): fix full-firmware build fixes #291
Fixes #291	

Changes:	
-Creating different object files for firmware or bootloader.

Does this change need to mentioned in CHANGELOG.md?	
no	

Requires testing	
no

Additional info:
Some files are shared for bootloader and firmware for example [`setup.c` for firmware](https://github.com/skycoin/hardware-wallet/blob/12968889ca8d5e96201e2defeb92e2c137911f73/tiny-firmware/Makefile#L7) and [`setup.c` for bootloader](https://github.com/skycoin/hardware-wallet/blob/12968889ca8d5e96201e2defeb92e2c137911f73/tiny-firmware/bootloader/Makefile#L5). Those files are build [... cut]
commit 6f9756a4 (2019-07-03): Merge pull request #293 from simelo/stdevAlDen_t291_address_generation_problem

fix full-firmware build fixes #291
closed 2019-07-03 by olemis