skycoin/hardware-wallet #295: Cleaning code in fsm.c and fsm_impl.c

RomanMilishchuk, 2019-07-01:
**Cleaning code in fsm.c and fsm_impl.c**

Refactoring of the files above to enhance readibility and productivity of code.

**Is your feature request related to a problem? Please describe.**
The main problem is that code is hard to read for beginners and some points are hard to understand.

**Describe the solution you'd like**
The problems, that are found are below in comments:

YuraYelisieiev, 2019-07-01:
While reviewing the code of [fsm_impl.c](https://github.com/skycoin/hardware-wallet/blob/12968889ca8d5e96201e2defeb92e2c137911f73/tiny-firmware/firmware/fsm.c), we found that there are a lot of defined functions there. Which increase the size of the binary because of every usage of the macros unfolds in the ASM, so if we are using macros often it enlarges our binary significantly. We must rewrite it into the function calls.

RomanMilishchuk, 2019-07-01:
fsm.c and fsm_impl.c are too oversized and can be divided into smaller files, which will decrease the time of compilation because a compiler will work only with files, that you changed.

In the next version will be integration also of Bitcoin, so natural division on parts, that are not connected to any currency (only to device itself) and functions, that are connected to Skycoin.

olemis, 2019-07-01:
thanks for creating this issue , this is not priority for  v1 though

stdevAlDen, 2019-07-01:
> While reviewing the code of [fsm_impl.c](https://github.com/skycoin/hardware-wallet/blob/12968889ca8d5e96201e2defeb92e2c137911f73/tiny-firmware/firmware/fsm.c), we found that there are a lot of [...]

@YuraYelisieiev it's ok but is important to check the macros to check out how this apply, for example this is a check pin:

https://github.com/skycoin/hardware-wallet/blob/12968889ca8d5e96201e2defeb92e2c137911f73/tiny-firmware/firmware/fsm_impl.h#L58-L61

initialized

https://github.com/skycoin/hardware-wallet/blob/12968889ca8d5e96201e2defeb92e2c137911f73/tiny-firmware/firmware/fsm_impl.h#L28-L39

init

https://github.com/skycoin/hardware-wallet/blob/12968889ca8d5e96201e2defeb92e2c137911f73/tiny-firmware/firmware/fsm_impl.h#L23-L28

so, can you find one that as function call is a better alternative?

SkycoinSynth, 2019-09-18:
Removing the macros may increase the binary size.

RomanMilishchuk, 2019-09-18:
@SkycoinSynth when you are using macro, then it on the time of preprocessing replaces the macro with the code (like inline functions), so actually using of macro increase the binary size, if they are used more then one time. Functions decrease the binary size but use more memory in the runtime, because of the stack frames.

PR #349 (2019-10-01): Wardady t295 fixing fsm
Fixes #295 

 Changes:	
-	Reduced the number of macros used in fsm.c & fsm_mpl.c by replacing them with functions. Moved functions related only to Skycoin cryptocurrency to fsm_skycoin.c&fsm_skycoin_impl. Reworked tests to correspond with these changes.

 Does this change need to mentioned in CHANGELOG.md?	
yes

 Requires testing	
no	

 Comments about testing , should you have some
closed 2019-10-01 by RomanMilishchuk