skycoin/hardware-wallet #299: Makefile recompiles everything

Retsediv, 2019-07-04:
**Describe the bug**

If we have already compiled bootloader(`make bootloader`) and firmware(`make firmware`), then when we are trying to `make full-firmware`, which should combine bootloader and firmware files together, it recompiles everything (both bootloader and firmware recompile here).

Also, if we have already compiled firmware and bootloader and trying to sign(`make sign`), then it recompiles firmware.

**Environment information:**

- OS: Linux(for sure), Windows(did not try), Mac OS X(probably)
- Platform: the equivalent of `uname -smr` or `uname -smrv` e.g. Darwin 17.4.0 x86_64
- Firmware Version: last dev commit

**Steps to Reproduce**
Steps to reproduce the behaviour:

with signing:
1. make clean && make clean-lib
2. make bootloader
3. make firmware
4. make sign -> see that firmware recompiles

------
with making full firmware:
1. make clean && make clean-lib
2. make bootloader
3. make firmware
4. make full-firmware -> see that bootloader and firmware recompiles

**Actual behaviour**
Bootloader and firmware recompile

**Expected behaviour**
It does not have to recompile those files because they were previously compiled.

**Possible implementation**
To my mind, there are 2 things to fix:
1. Dependencies in Makefile are slightly smashed up
2. Check compilation flags for files(since they are different for bootloader and firmware)

stdevAlDen, 2019-07-04:
> **Describe the bug**
>
> If we have already compiled bootloader(`make bootloader`) and firmware(`make firmware`), then when we are trying to `make full-firmware`, which should combine bootloader and [...]

I'm unable to reproduce from 6e80c965eb18d2ca876f071d0890e6d727112604.

`arm-none-eabi-gcc --version`
arm-none-eabi-gcc (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

`make --version`
GNU Make 4.2.1
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

`uname -srm`
Linux 4.19.0-1-amd64 x86_64

But please, you(@Retsediv) should [note this](https://github.com/skycoin/hardware-wallet/blob/6e80c965eb18d2ca876f071d0890e6d727112604/Makefile#L105).

Retsediv, 2019-07-04:
@stdevAlDen could you, please, share the log(on something like pastebin) of executing the followings commands:
1. `make clean && make clean-lib`
2. `make bootloader`
3. `make firmware`
4. `make full-firmware`

stdevAlDen, 2019-07-04:
> @stdevAlDen could you, please, share the log(on something like pastebin) of executing the followings commands:
> 
> 1. `make clean && make clean-lib`
> 2. `make bootloader`
> 3. `make firmware`
> 4. `make full-firmware`

@Retsediv hey good catch, with `make clean-lib` I get a different bug(that the one you already reported, but definitely there is a bug here). the log in a file.
[commands.log](https://github.com/skycoin/hardware-wallet/files/3360389/commands.log)

Retsediv, 2019-07-04:
@stdevAlDen so, as you can see in your log file, after `make full-firmware` execution it recompiles bootloader and firmware

stdevAlDen, 2019-07-04:
> @stdevAlDen so, as you can see in your log file, after `make full-firmware` execution it recompiles bootloader and firmware

@Retsediv I no think so, from line 2285 to 2410 I no see a rebuild.
Instead making:
`make clean-lib`
`make bootloader`
At least in my system I'm getting a cyclic rebuild (167-2124), can you reproduce this?

olemis, 2019-07-04:
> @Retsediv I no think so, from line 2285 to 2410 I no see a rebuild.

Closing this based on this comment , which I could reproduce .

Retsediv, 2019-07-07:
@stdevAlDen look at lines 2383-2385 (bootloader) and 2393-2396(firmware)

stdevAlDen, 2019-07-08:
> @stdevAlDen look at lines 2383-2385 (bootloader) and 2393-2396(firmware)

@Retsediv
This is not a "Makefile recompiles everything", is just a relink due to the rebuild of two files (due to [this](https://github.com/skycoin/hardware-wallet/blob/6e80c965eb18d2ca876f071d0890e6d727112604/Makefile#L105) as I previously [mention here](https://github.com/skycoin/hardware-wallet/issues/299#issuecomment-508546309)).

@olemis from my side the only thing relevant that I can reproduce is [this](https://github.com/skycoin/hardware-wallet/issues/299#issuecomment-508569103), @Retsediv can you?

Retsediv, 2019-07-08:
@stdevAlDen got it, thank you for your update

Yes, my fault. It does not recompile everything, but it does not have to rebuild those two files also, since we have already built firmware and bootloader (and we have appropriate files like skyfirmware.bin and bootloader.bin).  So, my idea here is that our Makefile is not written properly because it does not take care of dependencies (in this case `bootloader` and `firmware` for `full-firmware` target) which were already satisfied.

But that was only your side. I have just tried to compile on my machine. 
1. During signing (check linux_makefile_log(sign&full-firmware).txt) it actually recompiles everything. Could you, please, try this case on your machine? 
2. Without signing I have the situation pretty the as you have (check linux_makefile_log(only_full-firmware).txt)

[linux_makefile_log(only_full-firmware).txt](https://github.com/skycoin/hardware-wallet/files/3369358/linux_makefile_log.only_full-firmware.txt)
[linux_makefile_log(sign&full-firmware).txt](https://github.com/skycoin/hardware-wallet/files/3369359/linux_makefile_log.sign.full-firmware.txt)

Retsediv, 2019-07-17:
@olemis why did you close this issue? It's not solved yet

olemis, 2019-07-17:
> Yes, my fault. It does not recompile everything

closing issue considering this

olemis, 2019-07-17:
> @olemis why did you close this issue? It's not solved yet

as per your comments above ... what needs to be done ?

Retsediv, 2019-07-17:
please, read the whole comment

> During signing (check linux_makefile_log(sign&full-firmware).txt) it actually recompiles everything. Could you, please, try this case on your machine?

so, we need to fix recompilation during signing (I will change issue title)

olemis, 2019-07-17:
ok @Retsediv issue assigned to you .

closed 2019-09-17 by YuraYelisieiev