skycoin/hardware-wallet #302: Verify code coverage setup

stdevAlDen, 2019-07-09:
**Feature description**

Describe the feature

**Is your feature request related to a problem? Please describe.**
Just check that the [code coverage report](https://github.com/simelo/skycoin-hardware-wallet/tree/adeaaa695caeea8f35a86d212714bf31d76f9ef3#generating-tests-code-coverage) works as expected in osx

**Possible implementation**

Follow [the instructions](https://github.com/simelo/skycoin-hardware-wallet/tree/adeaaa695caeea8f35a86d212714bf31d76f9ef3#generating-tests-code-coverage).

YuraYelisieiev, 2019-07-23:
@stdevAlDen @olemis doesn`t work for OSX.

stdevAlDen, 2019-07-23:
@YuraYelisieiev did you install the [required tools](https://github.com/skycoin/hardware-wallet#generating-tests-code-coverage)?
what error are you having?

YuraYelisieiev, 2019-07-24:
> @YuraYelisieiev did you install the [required tools](https://github.com/skycoin/hardware-wallet#generating-tests-code-coverage)?
> what error are you having?

@stdevAlDen yeap, I followed the instructions, but I have this error. 
```make
Undefined symbols for architecture x86_64:
  "_llvm_gcda_emit_arcs", referenced from:
      ___llvm_gcov_writeout in test_skycoin_crypto.o```

YuraYelisieiev, 2019-09-02:
@stdevAlDen  review last comment please.

YuraYelisieiev, 2019-09-17:
skycoin-api is a library so we can`t test it using clang.

closed 2019-09-17 by YuraYelisieiev