skycoin/hardware-wallet #316: Bootloader issues

YuraYelisieiev, 2019-07-18:
In general: 
- We need to get rid of the static asserts in files. For example in storage.c.

[rng.h](https://github.com/skycoin/hardware-wallet/blob/ae2e6584224d18c8b80c00d21f1ddca34d9bbea7/tiny-firmware/rng.h)
- We need to change the include structure. Because now the way it compiles is not appropriate. 

signatures.c:
- why do need [SIG_OK](https://github.com/skycoin/hardware-wallet/blob/a2898d47139d0391166ced038257290916f662bb/tiny-firmware/bootloader/signatures.h#L23), [SIG_FAIL](https://github.com/skycoin/hardware-wallet/blob/a2898d47139d0391166ced038257290916f662bb/tiny-firmware/bootloader/signatures.h#L24) for signatures ok, if it just can return true or false
- writing the same code [thrice](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/bootloader/signatures.c#L99) isnt the best way. I think that better write a cycle.

bootloader.c:

- [Here](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/bootloader/bootloader.c#L74) button is checked after 100000 cycles of nop, while some of button is pressed. What is the sense of it, because user can quick press in such way, that bootloader won't see it. And in this time processor is literally doing nothing, so it is better to all this time just checking, if button was pressed

firmware_sign_split.py:

what is the sense of this file, if it this never used ?

usb.c:

- [Here](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/bootloader/usb.c#L646) is checking if new firmware is valid and then,if not, wiped from skywallet, but instead just wiped backup of meta, which is not important, because never used(in the next [line](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/bootloader/usb.c#L649)  it is replaced by new meta.
- Also multiple times are repeating lines from [here](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/bootloader/usb.c#L494) to [here](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/bootloader/usb.c#L509), so would be good to make function from them.
- Test of the functionality
[HERE](https://github.com/skycoin/hardware-wallet/blob/ae2e6584224d18c8b80c00d21f1ddca34d9bbea7/tiny-firmware/bootloader/usb.c#L390). 
1. Are you confident that this number is good enough to test the uniform distribution? [Link](https://github.com/skycoin/hardware-wallet/blob/ae2e6584224d18c8b80c00d21f1ddca34d9bbea7/tiny-firmware/bootloader/usb.c#L412)
2.  CPU status is not implemented yet [Link](https://github.com/skycoin/hardware-wallet/blob/ae2e6584224d18c8b80c00d21f1ddca34d9bbea7/tiny-firmware/bootloader/usb.c#L424) . Is it ok? 
3. Can you give me a detailed description of the USB test [HERE](https://github.com/skycoin/hardware-wallet/blob/ae2e6584224d18c8b80c00d21f1ddca34d9bbea7/tiny-firmware/bootloader/usb.c#L394)?  

Created by @RomanMilishchuk  and @YuraYelisieiev

jdknives, 2019-07-19:
Thanks for the review. Do I understand that correctly that questions 1-3 need to be answered for v1 and the bullet points above are improvement proposals or questions  for v2? @RomanMilishchuk @YuraYelisieiev

gz-c, 2019-07-19:
What is wrong with the `_Static_assert`?

olemis, 2019-07-21:
> What is wrong with the `_Static_assert`?

There is nothing wrong with them . They are there to ensure some preconditions are met at compile time . Removing those can lead to major issues.

YuraYelisieiev, 2019-07-22:
> > What is wrong with the `_Static_assert`?
> 
> There is nothing wrong with them . They are there to ensure some preconditions are met at compile time . Removing those can lead to major issues.

As I remember Asserts needs to be removed in the release because developers uses them only during the testing period.

gz-c, 2019-07-22:
these asserts should not be stripped out of releases. if any of these asserts fail the program should abort immediately

RomanMilishchuk, 2019-07-22:
@jdknives As for me, next problems should fixed for v1:
- > Here button is checked after 100000 cycles of nop, while some of button is pressed. What is the sense of it, because user can quick press in such way, that bootloader won't see it. And in this time processor is literally doing nothing, so it is better to all this time just checking, if button was pressed
- > Here is checking if new firmware is valid and then,if not, wiped from skywallet, but instead just wiped backup of meta, which is not important, because never used(in the next line it is replaced by new meta.
- And the questions 1 -3

olemis, 2019-07-25:
> these asserts should not be stripped out of releases. if any of these asserts fail the program should abort immediately

indeed , if assertions are not met code will fail to be compiled .

RomanMilishchuk, 2019-07-25:
@olemis this was written above and canceled a few days ago, could you reply to my last comment above?

stdevAlDen, 2019-07-26:
> In general:
> 
> * ~We need to get rid of the static asserts in files. For example in storage.c.~ (no, see [comments below](https://github.com/skycoin/hardware-wallet/issues/316#issuecomment-513570119) )
> 
> [rng.h](https://github.com/skycoin/hardware-wallet/blob/ae2e6584224d18c8b80c00d21f1ddca34d9bbea7/tiny-firmware/rng.h)
> 
> * We need to change the include structure. Because now the way it compiles is not appropriate.

related to [this](https://github.com/skycoin/hardware-wallet/issues/54)?

> signatures.c:
>
> * why do need [SIG_OK](https://github.com/skycoin/hardware-wallet/blob/a2898d47139d0391166ced038257290916f662bb/tiny-firmware/bootloader/signatures.h#L23), [...]

there is no `bool` data type in c99, no `true` no `false`, just integers.

> * writing the same code [thrice](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/bootloader/signatures.c#L99) isnt the best way. I think that better write a cycle.

please use a permanent link here

> bootloader.c:
>
> * [Here](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/bootloader/bootloader.c#L74) button is checked after 100000 cycles of nop, while some of button is [...]

maybe related to this https://github.com/skycoin/hardware-wallet/issues/27

> usb.c:
>
> * [Here](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/bootloader/usb.c#L646) is checking if new firmware is valid and then,if not, wiped from skywallet, but instead [...]

maybe this is unused, I can not find it from [here](https://github.com/skycoin/hardware-wallet-protob/blob/bd9f95c53447f97ac56f24a415c25d04680552de/protob/messages/messages.proto#L16-L58).

> 
> Created by @RomanMilishchuk and @YuraYelisieiev

@YuraYelisieiev & @RomanMilishchuk pls check out comments.

jdknives, 2019-07-26:
@stdevAlDen thanks for the comments. However it seems that the most important part, the questions 1-3 at the end were not answered. Would you mind doing that?

stdevAlDen, 2019-07-26:
> In general:
> 
> * ~We need to get rid of the static asserts in files. For example in storage.c.~ (no, see [comments below](https://github.com/skycoin/hardware-wallet/issues/316#issuecomment-513570119) )
> 
> [rng.h](https://github.com/skycoin/hardware-wallet/blob/ae2e6584224d18c8b80c00d21f1ddca34d9bbea7/tiny-firmware/rng.h)
> 
> * We need to change the include structure. Because now the way it compiles is not appropriate.
> 
> signatures.c:
> 
> * why do need [SIG_OK](https://github.com/skycoin/hardware-wallet/blob/a2898d47139d0391166ced038257290916f662bb/tiny-firmware/bootloader/signatures.h#L23), [SIG_FAIL](https://github.com/skycoin/hardware-wallet/blob/a2898d47139d0391166ced038257290916f662bb/tiny-firmware/bootloader/signatures.h#L24) for signatures ok, if it just can return true or false
> * writing the same code [thrice](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/bootloader/signatures.c#L99) isnt the best way. I think that better write a cycle.
> 
> bootloader.c:
> 
> * [Here](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/bootloader/bootloader.c#L74) button is checked after 100000 cycles of nop, while some of button is pressed. What is the sense of it, because user can quick press in such way, that bootloader won't see it. And in this time processor is literally doing nothing, so it is better to all this time just checking, if button was pressed
> 
> firmware_sign_split.py:
> 
> what is the sense of this file, if it this never used ?
> 
> usb.c:
> 
> * [Here](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/bootloader/usb.c#L646) is checking if new firmware is valid and then,if not, wiped from skywallet, but instead just wiped backup of meta, which is not important, because never used(in the next [line](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/bootloader/usb.c#L649)  it is replaced by new meta.
> * Also multiple times are repeating lines from [here](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/bootloader/usb.c#L494) to [here](https://github.com/skycoin/hardware-wallet/blob/develop/tiny-firmware/bootloader/usb.c#L509), so would be good to make function from them.
> * Test of the functionality
>   [HERE](https://github.com/skycoin/hardware-wallet/blob/ae2e6584224d18c8b80c00d21f1ddca34d9bbea7/tiny-firmware/bootloader/usb.c#L390).
> 
> 1. Are you confident that this number is good enough to test the uniform distribution? [Link](https://github.com/skycoin/hardware-wallet/blob/ae2e6584224d18c8b80c00d21f1ddca34d9bbea7/tiny-firmware/bootloader/usb.c#L412)
> 2. CPU status is not implemented yet [Link](https://github.com/skycoin/hardware-wallet/blob/ae2e6584224d18c8b80c00d21f1ddca34d9bbea7/tiny-firmware/bootloader/usb.c#L424) . Is it ok?
> 3. Can you give me a detailed description of the USB test [HERE](https://github.com/skycoin/hardware-wallet/blob/ae2e6584224d18c8b80c00d21f1ddca34d9bbea7/tiny-firmware/bootloader/usb.c#L394)?
> 

@jdknives sure.
I have no idea on how much this matter for the bootloader, but I found an important detail, this code does not exist in [the original project](https://github.com/trezor/trezor-mcu/blob/8af1f3643651e1cbee6b6196401f9c63e7cfc0f9/bootloader/usb.c#L120-L395). In addition consider my last comment here.
PD: investing more time I can come back with a better answer/solution but I'm fighting with [this](https://github.com/skycoin/hardware-wallet-go/issues/150) right now, please let me know if this have priority.

> Created by @RomanMilishchuk and @YuraYelisieiev

olemis, 2019-07-30:
in any case this is not a v1 issue