skycoin/hardware-wallet #318: make test fails with compilation error on macos

gz-c, 2019-07-22:
**Describe the bug**

```
$ make test
...
firmware/fsm.c:241:14: error: use of undeclared identifier 'FailureType_Failure_FirmwarePanic'; did you mean
      'FailureType_Failure_FirmwareError'?
        case FailureType_Failure_FirmwarePanic:
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
             FailureType_Failure_FirmwareError
/Users/steve/skycoin/hardware-wallet/tiny-firmware//protob/c/types.pb.h:31:5: note: 'FailureType_Failure_FirmwareError'
      declared here
    FailureType_Failure_FirmwareError = 99
    ^
firmware/fsm.c:241:14: error: duplicate case value 'FailureType_Failure_FirmwareError'
        case FailureType_Failure_FirmwarePanic:
             ^
firmware/fsm.c:235:14: note: previous case defined here
        case FailureType_Failure_FirmwareError:
             ^
...
`

**Environment information:**

- OS: macos

**Steps to Reproduce**
Steps to reproduce the behavior:

`make test` 

**Expected behavior**
Should compile

YuraYelisieiev, 2019-07-22:
@gz-c There was the same problem on Windows, did you pull the last version? Because now it works for me.

YuraYelisieiev, 2019-07-22:
@gz-c But there is another problem. If we enter `make full-firmware` and then `make test` there appears the next problem:
```bash
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [test_skycoin_crypto] Error 1
make: *** [test] Error 2
```
However, if firstly `make clean` and then `make test` everything is okay.

gz-c, 2019-07-22:
It is fixed in develop, thanks

gz-c, 2019-07-22:
I have this problem on latest `develop` branch

YuraYelisieiev, 2019-07-22:
> @gz-c But there is another problem. If we enter `make full-firmware` and then `make test` there appears the next problem:
> 
> ```shell
> clang: error: linker command failed with exit code 1 (use -v to see invocation)
> make[1]: *** [test_skycoin_crypto] Error 1
> make: *** [test] Error 2
> ```
> 
> However, if firstly `make clean` and then `make test` everything is okay.

@gz-c Re-checked. I have only this one on macOS.

olemis, 2019-07-22:
should be solved by just updating `hardware-wallet-protob` submodule to its `master` HEAD .

olemis, 2019-07-22:
> should be solved by just updating `hardware-wallet-protob` submodule to its `master` HEAD .

e.g. 

```sh
git -C tiny-firmware/protob checkout master
git -C tiny-firmware/protob fetch origin
git -C tiny-firmware/protob merge origin/master
```

olemis, 2019-07-22:
oh no , my mistake ...

olemis, 2019-07-22:
@gz-c looks like a merge issue or something like that in your working copy . It seems to be working for me on `osx`

olemis, 2019-07-22:
> firmware/fsm.c:241:14: error: use of undeclared identifier 'FailureType_Failure_FirmwarePanic'; did you mean
>       'FailureType_Failure_FirmwareError'?

wait ... this is the root cause , which should be solved by following [steps above](https://github.com/skycoin/hardware-wallet/issues/318#issuecomment-513871940) .

closed 2019-07-22 by olemis