skycoin/hardware-wallet #319: Run TRNG tests on new devices

jdknives, 2019-07-23:
Test the TRNG tests that are described here: https://github.com/skycoin/hardware-wallet#validate-the-trng on the new devices received and used for production. Report the results here please.

RomanMilishchuk, 2019-07-23:
When `skycoin-hw-cli getMixedEntropy --outFile stm32_rng_mixed_$i.dat --entropyBytes 1048576` is running after 83-84 % of bytes are getted cli gives error. Example of one of it (out of 8):
```bash
[2019-07-23T12:57:35+03:00] INFO [skycoin-hw-cli]: Getting mixed entropy from device
[2019-07-23T12:57:35+03:00] INFO [skywallet]: Saving entropy to stm32_rng_mixed_8.dat
[===================================================================================>                  ] 83.89% (879616/1048576)[2019-07-23T12:58:05+03:00]
ERROR [skywallet]: Error decoding device response as fails msg calling DecodeFailMsg with wrong message type: MessageType_Success
[2019-07-23T12:58:05+03:00] ERROR [skywallet]: calling DecodeFailMsg with wrong message type: MessageType_Success
[2019-07-23T12:58:05+03:00] ERROR [skycoin-hw-cli]: calling DecodeFailMsg with wrong message type: MessageType_Success
```

But for raw entropy everything works and now running for them tests. Publish as soon as it will finish.

jdknives, 2019-07-23:
@olemis @stdevAlDen have you encountered this before? 
@RomanMilishchuk you followed the instructions for building the firmware, I assume?

stdevAlDen, 2019-07-23:
> @olemis @stdevAlDen have you encountered this before?
> @RomanMilishchuk you followed the instructions for building the firmware, I assume?

@jdknives I was able to reproduce, so I'm working in...
@RomanMilishchuk unfortunately there is no an alternative path that you can follow, I'm going to solve this asap.

stdevAlDen, 2019-07-24:
tele

> @jdknives I was able to reproduce, so I'm working in...
> @RomanMilishchuk unfortunately there is no an alternative path that you can follow, I'm going to solve this asap.

@RomanMilishchuk & @jdknives Fixed in https://github.com/skycoin/hardware-wallet-go/pull/149
Please, let me know if this work for you.

RomanMilishchuk, 2019-07-24:
That PR helped, so [here](https://drive.google.com/open?id=1Q4g7Pqs5b3kvnqE1ShbNIkPZGQuEu2yP) are results of tests.

jdknives, 2019-07-24:
@olemis @stdevAlDen I assigned you for the review of the results.

jdknives, 2019-07-25:
@stdevAlDen @olemis please confirm, whether you can analyse the results of above test run.

stdevAlDen, 2019-07-25:
> @stdevAlDen @olemis please confirm, whether you can analyse the results of above test run.

@jdknives sorry for the delay in responding, I was looking for in the files and I was able to find an improvement from the [previous run](https://github.com/skycoin/hardware-wallet/pull/157#issue-266297643). But my criteria cannot be decisive since I am not an expert in mathematics with statistics background, so other reviews are welcome.
@RomanMilishchuk good work

olemis, 2019-07-25:
@stdevAlDen need to compare results of mixed vs raw entropy

stdevAlDen, 2019-07-25:
> @stdevAlDen need to compare results of mixed vs raw entropy

@olemis in some specific points this get better results and in others get worse results, but the general, but the general tendency is to improve.
pd: mixed against raw

jdknives, 2019-07-29:
@RomanMilishchuk what are the results of the other 2 devices that you ran the tests against?

olemis, 2019-07-29:
@RomanMilishchuk to be sure , the test suite must run in RA (i.e. resolve ambiguity) mode so that a final decision be made regarding `WEAK` , see [dieharder man page](https://www.systutorials.com/docs/linux/man/1-dieharder/) .

@stdevAlDen is this the right `man` page for the command in use ?

stdevAlDen, 2019-07-29:
> @RomanMilishchuk to be sure , the test suite must run in RA (i.e. resolve ambiguity) mode so that a final decision be made regarding `WEAK` , see [dieharder man page](https://www.systutorials.com/docs/linux/man/1-dieharder/) .
> 
> @stdevAlDen is this the right `man` page for the command in use ?

@olemis [done](https://github.com/skycoin/hardware-wallet/pull/325)

olemis, 2019-07-30:
@RomanMilishchuk could you please rerun TRNG test suite after changes committed in #325

RomanMilishchuk, 2019-07-30:
@olemis ok

jdknives, 2019-08-05:
@RomanMilishchuk what was the outcome of that?

RomanMilishchuk, 2019-08-06:
@jdknives I have discussed it in Discord, sending results into Telegram chat.

olemis, 2019-08-14:
> @RomanMilishchuk what was the outcome of that?

Results look very good

```sh
$ grep PASSED ./stm32_rng_raw_8.* | wc -l
     114
$ grep WEAK ./stm32_rng_raw_8.* | wc -l
       3
```

same for mixed

```sh

$ grep WEAK ./stm32_rng_mixed_8.* | wc -l
       3
$ grep PASSED ./stm32_rng_mixed_8.* | wc -l
     114
```

... even better than `/dev/urandom`

```sh
$ grep WEAK ./dev_urandom_8.dieharder | wc -l
       5
$ grep PASSED ./dev_urandom_8.dieharder | wc -l
     109
```

olemis, 2019-08-14:
hence I'll close this for now

closed 2019-08-14 by olemis