skycoin/hardware-wallet #321: The process for backing up the seed is not canceled correctly

Senyoret1, 2019-07-24:
**Describe the bug**
If the user is backing up the seed (the device is showing the words in the screen) and it is requested to cancel the operation, the operation is not canceled correctly. Instead of showing the home screen again, the device shows the confirmation screen asking the user if the seed was properly backed up and the device buttons do nothing, so the device appears to be unresponsive.

**Environment information:**

- OS: Windows 10
- Platform: X64
- Firmware Version: #78d0626

**Steps to Reproduce**
Steps to reproduce the behavior (with the hw wallet daemon):

1. Connect a wiped hw wallet and configure it with a random seed by calling `/generate_mnemonic` (https://github.com/skycoin/hardware-wallet-daemon/blob/master/src/api/README.md#generate-mnemonic).

2. Start backing up the seed by calling `/backup` (https://github.com/skycoin/hardware-wallet-daemon/blob/master/src/api/README.md#backup-seed)

3. While the device screen is still showing a word, cancel the operation by calling `/cancel` (https://github.com/skycoin/hardware-wallet-daemon/blob/master/src/api/README.md#cancel). The daemon will return an unexpected response:
```
{
    "error": {
        "message": "closed device",
        "code": 500
    }
}
```

4. Check the device screen. It should be asking for confirmation and the physical buttons will not respond.

5. Call `/cancel` again. This time the daemon will return the expected response:
```
{
    "data": "Action cancelled by user"
}
```

6. Check the device screen. It should be showing the Skycoin logo.

**Actual behavior**
When cancelling the operation, the device does not return to the home screen.

**Expected behavior**
When cancelling the operation, the device should return to the home screen.

**Additional context**
This is affecting the integration with the desktop wallet, because any operation must be cancelled after a timeout (the timeout is mandatory because it is imposed by web browsers).

It is possible to solve it just by calling `/cancel` 2 times, but it is a hacky workarround.

**Possible implementation**

stdevAlDen, 2019-07-25:
@Senyoret1 unable to reproduce with `firmware` and `skycoin-hw-cli`.
@olemis can you please try this?

stdevAlDen, 2019-07-25:
@Senyoret1 we found some errors, working on...

PR #322 (2019-08-02): Fix cancel backup ref #321
Fixes #321

Changes:	
- Change `reset_backup` signature to return an error.
- Check `reset_backup` ret value and go to home screen if non ok.
- Do not send cancel response from fsm layer.

Does this change need to mentioned in CHANGELOG.md?	
no	

Requires testing	
[later](https://github.com/skycoin/hardware-wallet/issues/178)
commit d43657c5 (2019-08-02): Merge pull request #322 from simelo/stdevAlDen_t321_fix_cancel_backup

Fix cancel backup fixes #321
closed 2019-08-02 by olemis