skycoin/hardware-wallet #327: Error in pub keys references in documentation.

stdevAlDen, 2019-08-06:
**Describe the bug**
Wrong documentation for pub keys location.

**Environment information:**

- Any

**Steps to Reproduce**
Steps to reproduce the behavior:

1. Go to [this link](https://github.com/skycoin/hardware-wallet/tree/d43657c574dad5ed5c823a32f30b04e94fd6f356/tiny-firmware#you-need-three-signatures).

**Actual behavior**
Wrong path documentation for pub keys.

**Expected behavior**
The right path is this

https://github.com/skycoin/hardware-wallet/blob/d43657c574dad5ed5c823a32f30b04e94fd6f356/Makefile#L12-L16

**Additional context**
Add a reference [from here](https://github.com/skycoin/hardware-wallet/tree/d43657c574dad5ed5c823a32f30b04e94fd6f356#sign-firmware) too.

SkycoinSynth, 2019-09-18:
>Steps to Reproduce

This is not a bug report. It is an issue with the source code.

Why are you writing a suggested change to source code as if it was a bug report?

jdknives, 2019-09-20:
Just fixed this issue. It is cosmetic.

closed 2019-09-20 by jdknives