skycoin/hardware-wallet #329: Openocd support

stdevAlDen, 2019-08-12:
**Feature description**

Confirm if we are going to support openocd support
https://github.com/skycoin/hardware-wallet/blob/d43657c574dad5ed5c823a32f30b04e94fd6f356/Makefile#L229-L230

if "yes", then check if this is working properly.

**Describe the solution you'd like**
Work based on the decision.

Macrosrider, 2019-08-14:
@stdevAlDen where is openocd.cfg file?

Macrosrider, 2019-08-16:
@stdevAlDen, I wrote a simple *.cfg file, which flash firmware to the hw (more info here bdba94f). "make oflash" command works perfect on Linux and Windows. Can someone check it on MacOS?

stdevStark, 2019-08-16:
> Can someone check it on MacOS?

I can take a look

stdevStark, 2019-08-16:
> I wrote a simple *.cfg file, which flash firmware to the hw (more info here [525d9a5](https://github.com/skycoin/hardware-wallet/commit/525d9a5b9a47bf9e41baafe3da5f74817642b897)).

could you please add a new PR ?

Macrosrider, 2019-08-16:
@stdevStark , will do it in a few minutes

stdevStark, 2019-08-16:
> more info here [525d9a5](https://github.com/skycoin/hardware-wallet/commit/525d9a5b9a47bf9e41baafe3da5f74817642b897)

@Macrosrider how can I `git fetch` those changes ? what fork is yours ?

Macrosrider, 2019-08-16:
@stdevStark, please check related PR. Suddenly, it failed on Travis and I have no idea why, because I've just fetch upstream of skycoin/develop

stdevAlDen, 2019-08-17:
> @stdevStark, please check related PR. Suddenly, it failed on Travis and I have no idea why, because I've just fetch upstream of skycoin/develop

@Macrosrider due to an update in the ubuntu repository, should be get solved after merging [this pr](https://github.com/skycoin/hardware-wallet/pull/333)
@olemis | @jdknives apply for [this](https://github.com/skycoin/hardware-wallet/pull/334) and all incoming pr, so please review -> merge asap.

PR #326 (2019-08-31): Windows support
Fixes #192 
Fixes #329 

 Changes:	
- Added emulator build support for Windows users

 Does this change need to mentioned in CHANGELOG.md?	
 no	

 Requires testing	
 no	

 Comments about testing , should you have some
 All instructions for setup are mentioned in tiny-firmware/README.md
commit fea50ff3 (2019-08-31): Merge pull request #326 from Macrosrider/windows_support

Windows support fixes #192 fixes #329
PR #335 (2019-08-31): Added OpenOCD config file, which flash firmware into hw
Fixes #329 	

 Changes:	
- Added OpenOCD config file, which flash firmware into hw	

 Does this change need to mentioned in CHANGELOG.md?	
yes | no	

 Requires testing	
yes, on Windows and Linux it works perfect, need to be tested on MacOS

 Comments about testing , should you have some
commit 491a76fe (2019-08-31): Merge pull request #335 from Macrosrider/openocd_support

Added OpenOCD config file, which flash firmware into hw fixes #329
closed 2019-08-31 by olemis