skycoin/hardware-wallet #350: Firmware build process does not work

jdknives, 2019-09-20:
**Describe the bug**
After installing the tools as described in the Readme on `master`, we tried to run `make sign`, `make firmware` and `make full-firmware`. The following error popped up: 

`make CC=arm-none-eabi-gcc AR=arm-none-eabi-ar -C tiny-firmware/vendor/skycoin-crypto/ libskycoin-crypto.a
make[1]: Entering directory '/home/mint/Desktop/hardware-wallet/skycoin-api'
arm-none-eabi-gcc -O3 -g -fPIC -std=gnu99 -W -Wall -Wextra -Wimplicit-function-declaration -Wredundant-decls -Wstrict-prototypes -Wundef -Wshadow -Wpointer-arith -Wformat -Wreturn-type -Wsign-compare -Wmultichar -Wformat-nonliteral -Winit-self -Wuninitialized -Wformat-security -Werror -I/home/mint/Desktop/hardware-wallet/skycoin-api/  -o skycoin_crypto.o -c skycoin_crypto.c
In file included from tools/sha2.h:35:0,
                 from skycoin_crypto.h:15,
                 from skycoin_crypto.c:12:
/usr/lib/gcc/arm-none-eabi/6.3.1/include/stdint.h:9:26: fatal error: stdint.h: No such file or directory
 # include_next <stdint.h>
                          ^
compilation terminated.
Makefile:54: recipe for target 'skycoin_crypto.o' failed
make[1]: *** [skycoin_crypto.o] Error 1
make[1]: Leaving directory '/home/mint/Desktop/hardware-wallet/skycoin-api'
Makefile:119: recipe for target 'arm-skycoin-crypto-lib' failed
make: *** [arm-skycoin-crypto-lib] Error 2
`

@RomanMilishchuk suggested running `apt install build-essential`, which I did, but the error still appeared. I did not experience those issues on Ubuntu. Since Synth is on Mint, we need to get it running there for the signature. 

**Environment information:**

- OS: Linux Mint 19.02

jdknives, 2019-09-21:
@YuraYelisieiev This is what I get while running `make sign`. (Make firmware was successful)

```
make[2]: Nothing to be done for 'all'.
  BUILD   lib/stm32/f1
make[2]: Nothing to be done for 'all'.
  BUILD   lib/stm32/f2
make[2]: Nothing to be done for 'all'.
  BUILD   lib/stm32/f3
make[2]: Nothing to be done for 'all'.
  BUILD   lib/stm32/f4
make[2]: Nothing to be done for 'all'.
  BUILD   lib/stm32/f7
make[2]: Nothing to be done for 'all'.
  BUILD   lib/stm32/l0
make[2]: Nothing to be done for 'all'.
  BUILD   lib/stm32/l1
make[2]: Nothing to be done for 'all'.
  BUILD   lib/stm32/l4
make[2]: Nothing to be done for 'all'.
  BUILD   lib/lpc13xx
make[2]: Nothing to be done for 'all'.
  BUILD   lib/lpc17xx
make[2]: Nothing to be done for 'all'.
  BUILD   lib/lpc43xx/m4
make[2]: Nothing to be done for 'all'.
  BUILD   lib/lpc43xx/m0
make[2]: Nothing to be done for 'all'.
  BUILD   lib/lm3s
make[2]: Nothing to be done for 'all'.
  BUILD   lib/lm4f
make[2]: Nothing to be done for 'all'.
  BUILD   lib/efm32/tg
make[2]: Nothing to be done for 'all'.
  BUILD   lib/efm32/g
make[2]: Nothing to be done for 'all'.
  BUILD   lib/efm32/lg
make[2]: Nothing to be done for 'all'.
  BUILD   lib/efm32/gg
make[2]: Nothing to be done for 'all'.
  BUILD   lib/sam/3a
make[2]: Nothing to be done for 'all'.
  BUILD   lib/sam/3n
make[2]: Nothing to be done for 'all'.
  BUILD   lib/sam/3s
make[2]: Nothing to be done for 'all'.
  BUILD   lib/sam/3u
make[2]: Nothing to be done for 'all'.
  BUILD   lib/sam/3x
make[2]: Nothing to be done for 'all'.
  BUILD   lib/sam/d
make[2]: Nothing to be done for 'all'.
  BUILD   lib/vf6xx
make[2]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/mint/hardware-wallet/tiny-firmware/vendor/libopencm3'
make FIRMWARE_SIGNATURE_PUB_KEY1=03d7fe879bea92c657797881cc2437fea86337ed4dce19859f43d023c1772c81dd FIRMWARE_SIGNATURE_PUB_KEY2=02ce15278b4d1cf5c20a2a518b95438bc1d497b1d55bddf69dbb253b538c131625 FIRMWARE_SIGNATURE_PUB_KEY3=02bebd3856b3fdadc54714220819946980d413edc8bc49f679b88fd2bf99ae0e15 FIRMWARE_SIGNATURE_PUB_KEY4=02015aa7da6acb423266a50db8821b42493cd2d874911c309209a003a1051c5d74 FIRMWARE_SIGNATURE_PUB_KEY5=03579beeeb075dcd20c63cbd3851ff0c81b1448c4f28597ce28924344a3f4bdbd4 REVERSE_BUTTONS=1 VERSION_MAJOR=1 VERSION_MINOR=7 VERSION_PATCH=0 GLOBAL_PATH=/home/mint/hardware-wallet/ -C tiny-firmware/ add_meta_header
make[1]: Entering directory '/home/mint/hardware-wallet/tiny-firmware'
make[1]: *** No rule to make target '/home/mint/hardware-wallet/tiny-firmware//vendor/skycoin-crypto/lib/libskycoin-crypto.a', needed by 'libskyfirmware.a'.  Stop.
make[1]: Leaving directory '/home/mint/hardware-wallet/tiny-firmware'
Makefile:170: recipe for target 'tiny-firmware/skyfirmware.bin' failed
make: *** [tiny-firmware/skyfirmware.bin] Error 2

```

YuraYelisieiev, 2019-09-22:
...he problem was that we were run 
git submodule update —init —recursive
on a develop and building binaries on master.

Steps to fix it on Mint 19, clean system.
```
1  sudo apt install git
    2  sudo apt install python3-pip
    3  sudo apt install gcc-arm-none-eabi
    4  git clone https://github.com/skycoin/hardware-wallet.git
    5  cd hardware-wallet/
    6  git checkout master
    7  git status
    8  git submodule update --init --recursive --remote
    9  make clean
   10  PYTHON=python3
   11  PYTHON=python3 make bootloader-mem-protect
   12  sudo apt install libnewlib-arm-none-eabi
   13  make clean
   14  PYTHON=python3 make bootloader-mem-protect
   15  PYTHON=python3 make firmware
   16  PYTHON=python3 make sign
   17  sudo apt install build-essential
   18  PYTHON=python3 make sign```

YuraYelisieiev, 2019-09-23:
@jdknives requested to separate instructions for the signing procedure and setup.
Setup:
```sudo apt install git
    2  sudo apt install python3-pip
    3  sudo apt install gcc-arm-none-eabi
    4  sudo apt install libnewlib-arm-none-eabi
    5  sudo apt install build-essential```
Sign:
```
git clone https://github.com/skycoin/hardware-wallet.git
    5  cd hardware-wallet/
    6  git checkout master
    8  git submodule update --init --recursive --remote
    9   make clean
   10  PYTHON=python3
   11  PYTHON=python3 make bootloader-mem-protect
   15  PYTHON=python3 make firmware
   16  PYTHON=python3 make sign
```

jdknives, 2019-09-23:
@YuraYelisieiev Thanks. Will give that a run.

jdknives, 2019-09-24:
Alright. Needed an apt update, but should be complete now. I verified that I got into the signing stages.

closed 2019-09-24 by jdknives